stelgenhof/AiLight #4: OTA doesn't seem to be working.

brahmafear, 2017-05-01:
Successfully flashed two bulbs using wired connection.  Bulbs are operating.  

Trying to upload via OTA in Atom with `pio run -t upload -e dev-ota  --upload-port <IP>` but getting a failure for no response:

```
18:36:46 [INFO]: Starting on 0.0.0.0:38558
18:36:46 [INFO]: Upload size: 401424
18:36:46 [INFO]: Sending invitation to: 192.168.9.228
Authenticating...OK
18:36:46 [INFO]: Waiting for device...
18:36:56 [ERROR]: No response from device
*** [upload] Error 1
```

stelgenhof, 2017-05-01:
There could be various reasons for this. One of them is that the if the OTA port in the firmware (set in config.h) differs from the one in the platformio.ini file. I experienced that myself a couple of times when testing the firmware :)

OTA updates are sometimes not immediately successful. Have you tried it again? Can you try also the 'prod-ota' environment?

brahmafear, 2017-05-01:
Tried on two bulbs - multiple times per bulb.  Since OTA auth is successful, I don't think it is the port - but I checked: both platform.io and config.h set to port 8266.  Same result on prod-ota.

stelgenhof, 2017-05-01:
Just tried a few times on a different device. Not all tries were successful but could upload it perfectly using the 'dev-ota' environment:

```
platformio run --target upload --environment dev-ota
Uploading .pioenvs/dev-ota/firmware.bin
08:40:28 [DEBUG]: Options: {'esp_ip': 'AiLight-CC4108.local', 'host_port': 52063, 'image': '.pioenvs/dev-ota/firmware.bin', 'hos
t_ip': '0.0.0.0', 'auth': 'hinotori', 'esp_port': 4266, 'spiffs': False, 'debug': True, 'progress': True}
08:40:28 [INFO]: Starting on 0.0.0.0:52063
08:40:28 [INFO]: Upload size: 401488
08:40:28 [INFO]: Sending invitation to: AiLight-CC4108.local
Authenticating...OK
08:40:28 [INFO]: Waiting for device...

Uploading: [============================================================] 100% Done...
 
08:40:37 [INFO]: Waiting for result...
08:40:37 [INFO]: Result: OK
========================= [SUCCESS] Took 39.25 seconds =========================
 
================================== [SUMMARY] ==================================
Environment dev         [SKIP]
Environment dev-ota     [SUCCESS]
Environment prod        [SKIP]
Environment prod-ota    [SKIP]
========================= [SUCCESS] Took 39.25 seconds =========================
```

brahmafear, 2017-05-02:
OK.  Will keep trying.  There was another person on the HomeAssistant gitter channel who reported problems with OTA as well.  Perhaps she/he will see this issue and comment.

stelgenhof, 2017-05-02:
Ok. You mean some people have reported issues with OTA in general, or with this AiLight firmware?

brahmafear, 2017-05-03:
Sorry for lateness of response:  HomeAssistant gitter channel reported problems with this firmware and OTA updates.  

I just tested with very latest pull from git.  Serial upload works great.  Firmware works great.  Still cannot get OTA working.  Here is dump from PIO console (I edited the auth pwd):

`pio run --target upload -e dev-ota --upload-port 192.168.9.244`

Bunch of compiling logs....

```
19:31:34 [DEBUG]: Options: {'esp_ip': '192.168.9.244', 'host_port': 47101, 'image': '.pioenvs/dev-ota/firmware.bin', 'host_ip
': '0.0.0.0', 'auth': 'abcabcabc', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
19:31:34 [INFO]: Starting on 0.0.0.0:47101
19:31:34 [INFO]: Upload size: 401424
19:31:34 [INFO]: Sending invitation to: 192.168.9.244
Authenticating...OK
19:31:34 [INFO]: Waiting for device...
19:31:44 [ERROR]: No response from device
*** [upload] Error 1
```

Any thoughts?  I've done successful OTA updates from Atom with other firmwares (espurna) so I don't think it is an environment issue - but not fully ruling that out.

stelgenhof, 2017-05-03:
Would you mind trying the 'prod-ota' environment? The size of the prod firmware build is somewhat smaller. I'm suspicious that the firmware size is getting too big for the ESP8266 in this light, which is only 1Mb...

Maybe I need to stop adding more features, haha

brahmafear, 2017-05-04:
Seems to be same result:
`pio run --target upload -e prod-ota --upload-port 192.168.9.244`

```
20:54:03 [DEBUG]: Options: {'esp_ip': '192.168.9.244', 'host_port': 19743, 'image': '.pioenvs/prod-ota/firmware.bin', 'host_i
p': '0.0.0.0', 'auth': 'abcabcabc', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
20:54:03 [INFO]: Starting on 0.0.0.0:19743
20:54:03 [INFO]: Upload size: 356960
20:54:03 [INFO]: Sending invitation to: 192.168.9.244
Authenticating...OK
20:54:03 [INFO]: Waiting for device...
20:54:13 [ERROR]: No response from device
*** [upload] Error 1
```

stelgenhof, 2017-05-04:
I'm having mixed results myself as well. I thought it was the sketch size, however after reducing some code (testing) I got same results. Need further exploration unfortunately...

stelgenhof, 2017-05-04:
Got much better results now! 
Can you comment out the 'extra_script' line in your platformio.ini file? 

I noticed when not using this, OTA works perfectly. Somehow the build script for the UI is interfering with the OTA upload. Some more investigation is needed. And probably another way to build the UI automatically.

brahmafear, 2017-05-04:
Hmm.  Commented out the "extra_script" line in both the dev-ota and prod-ota environments.  Saved.  Re-ran same pio command above against 2 bulbs.  Same result.  Auth successful.  Times out waiting for device.  Will keep experimenting.

brahmafear, 2017-05-04:
More info.  ESP module is crashing as soon as OTA starts.  Managed to catch this from serial port while attempting an OTA upgrade:
```
[OTA] Start

Exception (0):
epc1=0x40106f89 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0c60 end: 3fff0eb0 offset: 01a0

>>>stack>>>
3fff0e00:  3fffdad0 00000000 3fff1634 40210415  
3fff0e10:  3fff0e40 3ffef9d8 00000000 40202cc3  
3fff0e20:  3fffdad0 3fff34bc 00000000 402056bd  
3fff0e30:  3fffdad0 00000000 3ffef9d4 402042c8  
3fff0e40:  3fff1634 3ffefe90 00000000 3fffdad0  
3fff0e50:  4022495f 3fff0d3c 3ffe8620 3ffefe88  
3fff0e60:  3fffdad0 00000000 3fff1634 4021069e  
3fff0e70:  3fffdad0 00000000 3ffefe90 40213018  
3fff0e80:  00000000 00000000 00000001 40215e9c  
3fff0e90:  3fffdad0 00000000 3ffefe90 402027f6  
3fff0ea0:  feefeffe feefeffe feefeffe 40203ab4  
<<<stack<<<

```

stelgenhof, 2017-05-04:
To check if it was the AiLight or just the OTA piece, I started from the beginning and uploaded the BasicOTA sketch. I get a similar response:

```
Sketch uses 244707 bytes (23%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33384 bytes (40%) of dynamic memory, leaving 48536 bytes for local variables. Maximum is 81920 bytes.
python /home/sacha/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/espota.py -i 192.168.1.184 -p 8266 --auth=123 -f /tmp/arduino_build_948943/BasicOTA.ino.bin 
08:43:37 [ERROR]: No Answer
```
Funny thing is right after a code change and compilation, the upload via OTA works fine. However, if do it again (without any code change or whatsoever), I get the "No Response" message also.

Behaviour almost seems as if the binary code hasn't changed, nothing is happening. I saw the same when using PlatformIO.

stelgenhof, 2017-05-05:
Did some more experimenting. For the last couple of hours all my OTA uploads have been successful. 
I applied various changes:
- changing the code
- reboots
- power off
- different device
- different OTA password / host

Strangely all of them were successful and didn't get the "No answer" message.

The only time I got the "No Answer" message was when I tried to do an OTA upload quickly after another. I guess the device was not ready yet / restarting. If I waited about 10-15 seconds after the upload the subsequent one was always successful.

stelgenhof, 2017-05-05:
After some more experimenting I have made some minor changes to the OTA part of the firmware. Since then all my OTA uploads were successful. Would you mind checking again the latest code?

brahmafear, 2017-05-05:
Working!  Going to serial flash all my previous bulbs to make sure it works there too.  

_Thanks_ !

closed 2017-05-08 by stelgenhof