stelgenhof/AiLight #5: MQTT settings not saved

brahmafear, 2017-05-01:
Changes to MQTT topics made via web interface are not preserved across device reboots.

stelgenhof, 2017-05-01:
I haven't experienced this myself, to be honest. Let me test that again,

stelgenhof, 2017-05-01:
Just tested it myself and could change the MQTT successfully via the web interface. I restarted the device and also unplugged it from the power and restarted it. The changes were preserved.

Which version of the firmware are you using? I'm using the latest build (master).

brahmafear, 2017-05-01:
Hmm.  This is from github pull from middle of last week - right before you switched to async MQTT.  

Steps:
Plugged in / powered on bulb.  
Opened web interface.  
Hard refreshed page in browser.
Clicked Settings.
Changed state and command topics.
Clicked save.  
Unplugged light.
Plugged in / powered on bulb.  
Opened web interface.  
Hard refreshed page in browser.
Clicked Settings.
Old topics in place.  

Confirmed this by publishing to to the two different topics.

stelgenhof, 2017-05-01:
Hmm, strange. My initial code (not on Github) had some issues not preserving the data in EEPROM but that was a while ago. I tested the EEPROM features more than a dozen of times :)

Could you perhaps try the latest master branch from Github and see if that makes a difference?

brahmafear, 2017-05-04:
Seems to be resolved in 0.3.0.  Closing.  Thanks.

closed 2017-05-04 by brahmafear