stelgenhof/AiLight #7: OTA to latest version seems to have corrupted saved settings

brahmafear, 2017-05-09:
Head's up:  Upgrading from recent test builds to current seems to cause saved setting corruption.  Did the config_t structure change?  

After upgrading, bulb boots but cannot control LEDs.  Web UI doesn't show bulb info.  (See attached.)  Factory Reset does not clear settings.  

I hacked together the example EEPROM clear example with the BasicOTA example to get an OTA-able sketch that would wipe saved settings.  Uploaded that with Arduino IDE then used Pio to re-load AiLight.  That was enough to wipe the old data and access WebUI to re-enter settings.  

Same issue on two bulbs.  Coming from same previous firmware.  

[image]

stelgenhof, 2017-05-09:
The config_t struct has indeed been changed, but I haven't experienced any corruption while testing.

brahmafear, 2017-05-09:
Strange.  Watching serial console after upgrade, I saw successful connection to wifi, successful login to MQTT, successful subscribe to topic, then post to "@@@@@@@@@@@@@@@@@@@@@@@@" with garbage.

stelgenhof, 2017-05-09:
If the 'About' page shows no data, usually means the device is not ready yet (browser hasn't established connection with the device). If the device is ready then a refresh of your browser will solve that (Area for improvement :) )

Tested it again with another unit and had no issues at all. The garbage is the publish message to the new Last Will and Testament topic. Since it hasn't been configured yet with an upgrade it shows garbage like that. 

If you go to the settings page, the Status Topic field will likely be blank. Just enter the topic for your device (e.g. "AiLight09/status") and press Save. This will ensure the settings are saved correctly.

Perhaps I haven't made that clear in the instructions :(

brahmafear, 2017-05-09:
Don't think this was a browser refresh issue.  Waited for several minutes after ESP power on.  Also, Shift-refreshed page multiple times - as well as loaded in a new browser that hadn't accessed that bulb before.  

When About page was missing data, settings page didn't show any of the previously configured settings.  Light grey text in box instead describing each field.  I'll try to get a screenshot when I try upgrade next bulb.

brahmafear, 2017-05-09:
[image]

stelgenhof, 2017-05-09:
Yes, it's expected behavior :) the browser communicates with the firmware via WebSockets. If there is no connection established, then all information on the UI is not showing (All pages).

At the moment the firmware doesn't re-establish automatically the connection with the browser, if an update has been flashed.

brahmafear, 2017-05-09:
OK.  But if waiting / refreshing doesn't correct the issue??  
So far, the *only* way I've been able to restore functionality is to wipe eeprom.  

I've got three more bulbs on previous firmware (github version from May 5th).  Let me know if there is anything you'd like me to test while I upgrade them.  If not, I'll eeprom wipe, upgrade, and re-enter settings.  

Thanks.

stelgenhof, 2017-05-09:
I'm testing now as well. Upgrading from 0.2.0 to 0.3.0. I don't think an EEPROM wipe is needed.

brahmafear, 2017-05-09:
Hmm.  Maybe due to my bulbs were on a more recent version (May5th) instead of 0.2.0.

stelgenhof, 2017-05-09:
That might be very well possible as there have been quite some commits since 0.2.0. 
I did a clean upgrade from 0.2.0 to 0.3.0 and didn't get any corruption issues nor the UI showing blank data.

brahmafear, 2017-05-09:
All my bulbs updated now using eeprom wipe.  I'll mark issue to be closed as it seems to only be happening to me.  Thanks again.

closed 2017-05-09 by brahmafear

stelgenhof, 2017-05-09:
Thanks! Don't hesitate to open an issue if you experience something strange again :)

brahmafear, 2017-05-09:
Will do.  Are you open to code PRs?  I might try to add xy support so HomeAssistant color profiles can be used.

stelgenhof, 2017-05-09:
Of course! Happy to get PRs... XY support would be nice indeed.