stelgenhof/AiLight #10: Undefined Variable Errors Save Function

delian7, 2017-06-10:
Tried making a PR, but don't have permission. 

In line 449 in app.js, `var s` should be initialized as an empty object `var s ={}, msg = {};`. Otherwise an error ensues preventing saving any user settings. 

[image]

stelgenhof, 2017-06-11:
Thanks for catching this! I can confirm I also get this error message. Perhaps my JavaScript knowledge is getting rusty; I thought `var s, msg = {};` would initialize both variables. :( 

Already checked that your suggestions works. I'll change it asap.

As for the PR, did you create a fork and from there created a PR?

delian7, 2017-06-11:
Ahh I didn't fork: I tried cloning the repo in sourcetree, made a branch, and tried to push that branch to do a PR. That makes sense.

closed 2017-06-11 by stelgenhof