stelgenhof/AiLight #11: Cannot accept empty wifi password?

robbintt, 2017-06-26:
Hi, it looks like there is no way to use a wifi network that has no password. I believe it is due to the validation in the "settings" template.

I am not sure if I can give a password-less network in the firmware, I will try.

robbintt, 2017-06-26:
I have tested adding a wifi network to the firmware with no password - so far I cannot access the hostname over HTTP or see mqtt client, although the mqtt may just not be configured correctly in Home Automation, as I have no reference to check against yet.

nickrout, 2017-06-26:
I am not sure this is a real world problem.

robbintt, 2017-06-26:
An important part of software is recognizing that your use cases are not the only ones.

stelgenhof, 2017-06-26:
Although not recommended to use a WiFi network without a password, it still is a valid scenario.

I think the code changes will be minimal to alllow for WiFi without passwords. Let me have a look to make the necessary changes.

stelgenhof, 2017-06-26:
@robbintt I have a made code change so that WiFi networks without a password work as well. You can now leave the password field blank, without getting a validation message. 

I committed in the main master branch, so if you could try it out that would be great!

stelgenhof, 2017-06-30:
Closing this as it has been addressed in release v0.4.1.

closed 2017-06-30 by stelgenhof