stelgenhof/AiLight #12: Firmware Uploads - No change to bulb

robbintt, 2017-07-01:
I'm having an interesting issue with the AiLight bulb. I am able to successfully upload firmware to the bulb according to PlatformIO but it does not change the bulb's behavior. 

I have deleted my AiLight repository and the AiLight github repository and rebuilt everything from scratch to no avail.

I am getting a blue light with intermittent brief flashing when turning on the light bulb, regardless of the 'default' state that I specify in my `config.h` file.

I have ordered another bulb to test against as well. 

I am wondering if there is a section of the esp8266 flash that PlatformIO is not writing to that is somehow corrupted.

Is there a simple way to pull the bin back off the esp8266 so that I can compare it to the compiled file?

Verbose output of successful upload:
```
opening bootloader                                                                                                                    [32/1885]
resetting board
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
espcomm_send_command: receiving 2 bytes of data
Uploading 395808 bytes from .pioenvs/dev/firmware.bin to flash at 0x00000000
erasing flash
size: 060a20 address: 000000
first_sector_index: 0
total_sector_count: 97
head_sector_count: 16
adjusted_sector_count: 81
erase_size: 051000
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
setting timeout 15000
setting timeout 100
espcomm_send_command: receiving 2 bytes of data
writing flash
................................................................................ [ 20% ]
................................................................................ [ 41% ]
................................................................................ [ 62% ]
................................................................................ [ 82% ]
...................................................................              [ 100% ]
starting app without reboot
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
closing bootloader
======================================================== [SUCCESS] Took 52.26 seconds ========================================================
```

brahmafear, 2017-07-01:
I've also seen this issue. I resolved by uploading the eeprom clear sketch
from the eeprom library. Run that to zero out eeprom. Then re-upload the
AiLight firmware.

Hope this helps.

stelgenhof, 2017-07-02:
Hmm, strange. I've flashed the firmware to a dozen different bulbs but never seen this to be honest. 
@robbintt Would you mind sharing the exact steps you took, so I can have a better idea?

Thanks! Sacha

robbintt, 2017-07-02:
Thanks for the replies - I'll report back as soon as I can make some time.

stelgenhof, 2017-07-19:
@robbintt Any updates on this? Is it something similar to issue #13?

stelgenhof, 2017-08-29:
Closing this issue as I believe it has been resolved (See issue #13). Should it occur again, please don't hesitate to create a new issue.

closed 2017-08-29 by stelgenhof