stelgenhof/AiLight #15: Domoticz

duelago, 2017-08-04:
I really like this project. I'm up and running and everything works as advertised. 
I'm new to MQTT and my backend is Domoticz. Any plans to support Domoticz in a future release?

I really don't have the skills to implement this support myself.

Keep up the good work :)

 /David

stelgenhof, 2017-08-04:
Thanks for the kind words! Domoticz wasn't really on the radar but definitely can have a look into it.

stelgenhof, 2017-08-07:
I took some time to check Domoticz out and the MQTT integration is pretty simple. I have now a sandbox Domoticz and tried to set up a light/switch but coming from Home Assistant, the entire setup feels a bit strange. 

I managed successfully to publish an MQTT payload to Domoticz, but had a hard time figuring out how to integrate with an RGBW light. 

Do you have any pointers how I can do that? Guess I need to play around a little bit more..

stelgenhof, 2017-08-09:
After fiddling around a bit more in Domoticz, I got it working. Meaning: I could send and receive commands via MQTT. The way Domoticz handles this, is quite different from what I'm used to. 

Let me see how this could be integrated in the AiLight firmware.

closed 2020-03-05 by github-actions[bot]