stelgenhof/AiLight #16: Use an input type='color' instead of sliders?

bofh69, 2017-09-09:
Something like:
```html
          </tr>
            <th>Colour</th>
            <td><input class="" type="color" id="colour_lamp"></td>
          </tr>
```

```js
let colour_lamp = document.getElementById("colour_lamp");
colour_lamp.addEventListener("change", sendRGB2.bind(this), { passive: true});
colour_lamp.addEventListener("input", sendRGB2.bind(this), { passive: true});

function sendRGB2() {
  let msg = {
    'state': S_ON
  };

  msg[K_C] = {};
  msg[K_C][K_R] = parseInt(colour_lamp.value.substring(1,3), 16);
  msg[K_C][K_G] = parseInt(colour_lamp.value.substring(3,5), 16);
  msg[K_C][K_B] = parseInt(colour_lamp.value.substring(5,7), 16);

  websock.send(JSON.stringify(msg));
}
```
If the input event sends the message, perhaps an extra parameter should be sent as well so the
config isn't saved in that case (it would wear down the EEPROM much faster).

A drawback is that the lamp doesn't match the RGB colour of the screen. Some colour correction formula is needed to make this really good.

stelgenhof, 2017-09-10:
Thanks for the suggestion. Something similar I had in mind as well to do, but haven't gotten around it yet.

stelgenhof, 2017-09-13:
After having a look at your example, the input type='color' is not support by all browsers. Specifically
, iOS Safari has no support for it. Personally I use my iPhone a lot controlling my lights, so this wouldn't work in that occasion.

bofh69, 2017-09-13:
I noticed that now as well. I added a polyfill for it, but the one I use (spectrum) does not update its colour when the value is updated and since it comes from a CDN it doesn't work before the lamp is added to an AP.

stelgenhof, 2017-09-13:
In the beginning when creating the UI, I was also using external (CDN) libraries. That indeed cause the UI not to work if it is in AP mode. Therefor I decided to make everything embedded and use plain vanilla JS/CSS.

stelgenhof, 2017-09-25:
@bofh69 Just wanted to check if you still want to keep this issue/request? Otherwise, I will close it. Thanks!

stelgenhof, 2017-10-05:
@bofh69 Thanks! I would also like to use it if it would be more widely supported. I'll close it for now and if things changes we can address it again.

closed 2017-10-05 by stelgenhof