stelgenhof/AiLight #17: Add support for DIY devices

fonix232, 2017-09-09:
It would be great if AiLight supported homemade solutions too - since they are usually cheaper than an actual bulb, and are easily put together (say, a Wemos D1 Mini @ 2.5$ and an WS2812B LED ring for 3$).

The platform is technically already supported, but it isn't too configurable regarding the lights - I understand that the main goal is to support the official LED bulbs, but it would be awesome if there was a way to configure other, drop-in style controllers (e.g. the WS2811/12 NeoPixel LEDs), something similar to how Mongoose OS solved libraries (libraries can define their own settings, which is then used to generate the settings UI).

I understand that this is a huge thing, since there's no advanced features like interfaces (which would allow for an IoC-like selection of current LED provider), but it would be a welcome change nonetheless.

stelgenhof, 2017-09-10:
Thank you for the suggestion. I agree that would be great however this firmware was specifically made for LED bulbs with the MY9291 LED driver. 

Wouldn't it be better idea to use other firmwares like Espurna or Tasmota?

fonix232, 2017-09-10:
I did look at the other firmwares, but so far AiLight seems to be the closest to the project I plan.

The web UI in itself is a big plus, and the developed features too. ESPurna might work out, but as I saw it is more oriented at smart switches, not RGB LED lights.

stelgenhof, 2017-09-10:
Curious about your project, would you mind to share?

The main reason for me to create this UI is because I wanted to have a UI that is very user friendly, easy to maintain and of course good looking :) I believe the packaging is as much important (or even more) as the present itself.

ESPurna has support for a couple of different smart RGBW LED lights, not only smart switches. Recently there have been quite a few updates.

fonix232, 2017-09-10:
My project is basically making multiple Wemos D1 Mini controlled lights, possibly even having some manufactured - since the WS2812 LED rings are quite cheap (although they do not provide white LEDs), and around 3$ mass manufactured, one could have awesome WiFi-enabled LED rings around.

stelgenhof, 2017-09-14:
Sounds like a nice project. Have you considered then making your own PCB? With the ESP8266 chip?

Having thought about your suggestion some more, and although I think it could work, it would deviate quite far from the goal of this firmware. Actually Espurna supports WS2812 type LED's so that could be a good alternative. I think if adding support for other type of LED's, this firmware will be looking like Espurna (maybe except for the UI part) :)

fonix232, 2017-09-19:
Custom PCB is planned, yes. My ultimate goal is to create a company around this, with a fully custom ESP8266/ESP32 based firmware that is open source and community-driven, with various products based on it - both a hacked-oriented lineup where extra pins are exposed, and a "final", customer-oriented product that "just works".

stelgenhof, 2017-09-19:
Myself also planning on making some custom PCB's, just haven't got the time to do it yet. 
Might be challenging to make good and unique product, as there already so many ideas/hacks based on the ESP8266 out there.

fonix232, 2017-09-19:
Lots of hacks and ideas, yes, but not much that are well organized and not just a bunch of hacky code that works or not... A good product that actually works well is hard to achieve but I see some sales potential, especially if we can hit a good price/value ratio. Say, a 15$ LED light bulb that works well and can emulate even a Philips Hue lamp.

stelgenhof, 2017-09-19:
The primary reason for creating this AiLight firmware was to have a Web UI that is very user friendly and intuitive. A lot of people focus on the hardware/firmware part in their projects and leave the UX/UI at the last. UX/UI is crucial for the success of a product.

I think AiLight already made a step in the right direction, although I think it can still be improved much more :)

kmickeletto, 2018-01-03:
Do you think this would work for the Tyua smart sockets?  They are a simple on/off device with no color options.  I know they use the ESP8266 chipset.  So if the extra hardware isn't present, will this firmware just work with the on/off functions and wifi?

brahmafear, 2018-01-03:
@kmickeletto I use [espurna](https://bitbucket.org/xoseperez/espurna) for similar sockets.  But I like AiLight better for the bulbs.

packetinspector, 2018-01-12:
I've built some custom lights as well, I used this [code](https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs).  No front end, but has all the features you need for lights via MQ.