stelgenhof/AiLight #19: Support for color temperature

helderpe, 2017-09-18:
Ideally, these bulbs would adjust the rgb led and the white led to accurately reproduce white color temperatures. For example, I use the Yeelight RGBW bulbs with homeassistant. When selecting a color temperature, the bulb changes it's leds to the correct colors. When I do this in homeassistant with the AI-Light firmware, if I have the White LED set to maximum, it's hard to see any change in the color temperature. There seems to be some literature around the internet explaining hot to get this effect correctly (for example, http://planetpixelemporium.com/tutorialpages/light.html)

I can provide support for this too, by testing or trying to develop something. Any thoughts?

stelgenhof, 2017-09-18:
Currently when using the color temperature feature of AiLight, it only sets the RGB LED's of the Ai-Thinker light bulb. The White LED's operate independently from the RGB LED's.

I am familiar with that source you are referring to and have done quite some research about color temperature. As far as I am aware the current way it is implemented, works quite fine. Are you saying that the White LED's should be controlled in tandem with the RGB ones?

Happy to see your implementation/PR of this of course! :)

helderpe, 2017-09-18:
From what I understand, I think that the white LED should also be adjusted to produce the right color temperature. I'll take a shot at this (not really sure when I'll be able to do) and I'll let you know. 

Btw, I guess that the white LEDs are in the 4000k range, am I right?

helderpe, 2017-09-18:
Some additional information might be found here: http://blog.saikoled.com/post/44677718712/how-to-convert-from-hsi-to-rgb-white

stelgenhof, 2017-09-18:
Thanks for having a go at this. Would be cool! 
Don't know if the White LED's are in the 4000K range, unfortunately.

helderpe, 2017-09-18:
I did some additional research, and I found a simple rgb to rgbw conversion formula (derived from http://www.mirlab.org/conference_papers/International_Conference/ICASSP%202014/papers/p1214-lee.pdf):

M = max(Ri,Gi,Bi)
m = min(Ri,Gi,Bi)

Wo = if (m/M < 0.5) use ( (m*M) / (M-m) ) else M 
Q = 255
K = (Wo + M) / m
Ro = floor( [ ( K * Ri ) - Wo ] / Q )
Go = floor( [ ( K * Gi ) - Wo ] / Q )
Bo = floor( [ ( K * Bi ) - Wo ] / Q )

The stack overflow post can be found from here: https://stackoverflow.com/questions/21117842/converting-an-rgbw-color-to-a-standard-rgb-hsb-representation

This made me think that maybe we can adjust AiLightClass::setRGBW() to take this into consideration, which means that we would also be using the white LEDs to produce normal colors. I'll give this a try as soon as I can and see if the results are interesting.

helderpe, 2017-09-19:
Ok, so I modified the code, you can find it here: https://pastebin.com/jvvnaJhH

I basically call this function first thing on setRGBW();

The code compiles, and I was able to do an OTA, but it never finished the OTA process and I had to reflash the bulb manually with a USB programmer. Any thoughts?

stelgenhof, 2017-09-19:
Looking good!

OTA is known to be finicky sometimes. Have you tried again? Also were you able to upload via OTA before the code change?

helderpe, 2017-09-20:
It was the first time I did an OTA with this firmware. It basically got stuck, the bulb didn't connect to Wifi after the failed attempt (the firmware was fully uploaded to the bulb, but for some reason it didn't finished writing the firmware).

stelgenhof, 2017-09-20:
Have you tried again using OTA? Are you sure the firmware wasn't written successfully?

I would first try to upload via the USB/Serial Adapter to ensure the firmware works properly. After that you can try OTA.

helderpe, 2017-09-21:
Well, to upload the firmware via USB/Serial Adapter I had to always solder some cables to the lamp, Finally, yesterday I was able to 3D print a fixture that will make life easier. However, I'm still waiting for the pogo pins :(

stelgenhof, 2017-09-21:
Yes, the very first time you need to upload firmware via a USB/UART adapter. Same as you, I made a jig a while ago  using pogo pins to avoid any soldering. If you're interested, you can find some more details here: https://github.com/stelgenhof/AiLight/wiki/Connection

I've made about a dozen so far for others in the community :)

stelgenhof, 2017-09-30:
@helderpe I have tried your code and it caused an Exception (WDT Reset) on my AiThinker Light. Have you tried your code yourself as well? If so, do you see any significant difference?

After reading the paper and sources you've provided, I am still not sure if this is the common practice. Currently the White LED's in AiLight firmware are controlled independently from the RGB one's. 

Based on my own research, I am not sure whether the White LED's should always be controlled together with the RGB LED's or separately.

helderpe, 2017-09-30:
Hi @stelgenhof. I was unable to do any further tests, but that might explain why I was unable to complete the OTA upgrade. 
I guess the only way to have the color temperature that I was expecting is probably by having two sets of white LEDs, one for really warm temperatures, and other for the opposite (I think the Sonoff B1 has this kind of configuration). Thanks for the help.

stelgenhof, 2017-10-01:
@helderpe Thanks! Would you still like to proceed with this? As said, I am not sure what is the common practice. 

I'm more than happy to apply this change (and sort out the exception) if it is indeed better to include the White LED's as part of the colour temperature setting.

stelgenhof, 2017-11-23:
Closing this issue. Should you feel this issue to be addressed again, please reopen it or create a new one.

closed 2017-11-23 by stelgenhof