stelgenhof/AiLight #20: homebridge plugin AiLight

ageorgios, 2017-09-20:
Are there any plans for a homebridge plugin?

stelgenhof, 2017-09-20:
Not sure what you mean by a 'Homebridge plugin', but you can use Homebridge together with Home Assistant using the homebridge-homeassistant plugin.

I have used sometime ago, and was perfectly able to use iOS' Siri to control my light :)

badgerhome, 2017-09-20:
I use homeassistant-homebridge with AiLight MQTT bulbs, brightness and colours both work too!

ageorgios, 2017-09-21:
in case you don't use homeassistant there are plugins for specific devices, like homebridge-aqara (for xiaomi devices) homebridge-yeelight (for xiaomi bulbs) etc.

stelgenhof, 2017-09-21:
Ah, I see what you mean. Unfortunately I haven't really looked at that possibility yet :)

stelgenhof, 2017-10-01:
@ageorgios Your wish has been granted! Created a homebridge plugin for the AiLight firmware. More details you can find here: https://www.npmjs.com/package/homebridge-ailight or here: https://github.com/stelgenhof/homebridge-ailight

ageorgios, 2017-10-01:
you rock!

stelgenhof, 2017-10-02:
Closing this issue as a dedicated Homebridge plugin is now available here: https://github.com/stelgenhof/homebridge-ailight

closed 2017-10-02 by stelgenhof