stelgenhof/AiLight #23: Gamma boolean can not be set via REST or mqtt

petefats, 2017-10-13:
Thanks for the incredible firmware.  Its been an integral part of an outdoor project of mine using 24 lights.  After a recent snowstorm, part of the lights report gamma correction true, part of them report false.  I believe this is a EEPROM problem due to the cold, nothing to do with AiLight itself.  

The problem I'm running into however is that AiLight doesn't expose this variable to either REST or mqtt, making it difficult to return all of the lights the same state.  I'm uncertain if this is a bug or simply unimplemented (and working as expected).  I've attached a repro in case it helps.

[gamma-repro.txt](https://github.com/stelgenhof/AiLight/files/1381214/gamma-repro.txt)

stelgenhof, 2017-10-13:
Thanks! @petefats. In this case it is just a matter of 'unimplemented'. The firmware was targeted at HomeAssistant, which doesn't know anything about 'Gamma Correction'. 

I agree that it would be better to be able to set Gamma Correction via REST or MQTT as well. Let me see how that can be done :)

petefats, 2017-10-13:
Thanks for looking.  In the interim, I found that I could connect directly via websocket to adjust this variable.

stelgenhof, 2018-04-24:
@petefats I know you were able to work around this by using the Websockets. However, I have added the ability to set the Gamma Correction via REST API and MQTT now as well. 

It's not available yet as part of an official release, however you can try it out by compiling the firmware (develop branch) yourself.

Cheers! Sacha

stelgenhof, 2019-04-08:
Included in the just released version 0.6.0.

closed 2019-04-08 by stelgenhof