stelgenhof/AiLight #24: Upon power on, light turns on.

tsaitsai, 2017-10-22:
Sacha, thanks for this project.  Really useful.
I would like the option to have the bulb turn on upon power-on.  Right now, it seems bulbs stay off until explicitly given a command to turn on (so you have to monitor the mqtt status topic for that bulb).  If the bulbs defaults to turning on when powered, at some setting defined in the config maybe, then the bulb at least functions like a regular light bulb in the absence of a controller on the network.

stelgenhof, 2017-10-22:
@tsaitsai Not sure why you closed it, but I think this is indeed a good feature to have. Currently, the firmware sets the light to the last known state. So if it was on, it will be on again after being powered on again.

tsaitsai, 2017-10-22:
@stelgenhof I got confused!  I didn't realize I was testing different states - the older bulb was off when I power cycled and the newer bulb was on when power cycled, making me think you've already implemented this.

But yes, I think making it behave like a normal bulb in case of automation failure would be nice.  This would still work with physical light switches.

rbu, 2018-02-22:
I have to second this request. If you use this bulb in old school lamps with wall switches, it's extremely useful if they can default to a "dumb" behavior for users who do not use an app.

peterjohnston, 2018-04-07:
Thanks very much for this super-useful project! 

I have a further suggestion for this, in that each time the bulb power-cycles, it moves to a different state. Perhaps this could be part of a future "effects" enhancement where it cycles around them. 

This would allow a user to have the bulb move from, for example: dim, medium, bright, etc, on subsequent power-ups. 

This would give an instant change rather than having to wait for a wifi connection, report its existing state via MQTT then be told what its next state should be remotely.

stelgenhof, 2018-04-24:
@peterjohnston My apologies for the late reply! I have just added a feature allows you to set the behaviour of the light after a power cycle. As for your suggestion, would you mind creating a separate issue/feature request?

stelgenhof, 2018-04-24:
@tsaitsai @rbu Feature is done! It has been added to the develop branch, which means you have to compile the firmware yourself. Official release/binary will come (if it has become stable).

Let me know if you experience any issues.

Cheers! Sacha

stelgenhof, 2019-04-08:
Included in the just released version 0.6.0.

closed 2019-04-08 by stelgenhof