stelgenhof/AiLight #25: What happens if the Network changes? Any way to re-enter AP mode / Reset

kaiomatico, 2018-01-10:
Hey, first thank you for the awesome project! I ordered 2 bulbs which should arrive shortly, but I have a concern or question, as I can't find it documented anywhere in the Wiki:
-What happens if I get a new router with a different SSID and password or what happens if I configure the lights in the AP mode for the first time and enter the wrong Wi-Fi credentials? My guess is I can't enter the lights in the AP mode for correction again until I have to re-flash the firmware, am I right? This could be solved or implemented using maybe a timeout which will check if there is no Wi-Fi Connection for like 24 hours while being powered on and then reset to factory defaults (just like the button in the web interface does)

Tanks in advance! 👍  maybe this could be seen as a feature and could be implemented in the compile-time as option.

stelgenhof, 2018-01-10:
If a WiFi connection cannot be established (either first time connect or new WiFi network), then the firmware will change into AP mode allowing you to enter the required settings.

kaiomatico, 2018-01-10:
Perfect thank you.

closed 2018-01-10 by kaiomatico

xbmcnut, 2018-05-08:
What's the WiFi password in AP mode?