stelgenhof/AiLight #26: Arilux Bulb?

packetinspector, 2018-01-10:
I flashed the code on to one of these, but it just caused boot loops.

http://tinkerman.cat/yet-another-wifi-light-bulb/#more-2127

I'm going to try Espurna, but is there anything I might be able to change in your code to make it compatible?

stelgenhof, 2018-01-10:
I am not aware these Arilux can work or not with the AiLight firmware. According to that link, it uses the same LED driver as the Ai-Thinker Light, so from that it looks like it should work.

Unfortunately I don't have an Arilux bulb, so cannot really test if it works or not. (https://github.com/stelgenhof/AiLight/wiki/Compatible-Light-Bulbs)

Espurna might work in this case, as this a more generic firmware designed for general ESP8266 based devices.

packetinspector, 2018-01-11:
I suspect it has something to do with the flash/eeprom config.

stelgenhof, 2018-01-11:
Yes, that's possible. Also not all the ESP8266 chips used in these bulbs are the same. Always hard to assess which bulb works and which doesn't unfortunately.

stelgenhof, 2018-02-11:
@packetinspector Perhaps not related to your Arilux bulb, many have actually experienced reset loops (Software WDT reset) as well (including myself). I have made some fixes in the master branch which makes the firmware now work with Arduino ESP8266 v2.4.0 framework also.

Could you give it a try and let me know if it works or not?

Cheers! Sacha

stelgenhof, 2018-03-26:
@packetinspector Just wanted to follow up if you had any success with the latest master branch code?

xbmcnut, 2018-05-08:
I have just tried v5.13.1 on an Arilux bulb and while it flashed, the webpage does not load correctly preventing access to program any settings (WiFi, MQTT etc.)

 
[image]

stelgenhof, 2018-05-08:
@xbmcnut Sorry I don't recognize this screenshot and neither the version number you are referring to (v5.13.1). What is this firmware you are trying to flash?

xbmcnut, 2018-05-08:
Disregard. Seems I was connecting to another ESP unit running a MiLight bridge! Sorry about that. Don't know why that was up on 192.168.4.1 as it had been configured for WiFi already. My Arilux bulb does not seem to be working after flashing 0.5.0 release. No AP mode.

stelgenhof, 2018-05-08:
Can't really tell what the issue is you are experiencing as I don't have an Arilux bulb myself. Not confirmed if the firmware works with this lightbulb or not. Best is to check the output of the Serial connector to see if there are any clues why it is not working.

xbmcnut, 2018-05-08:
FYI, I tried espurna-1.12.6-arilux-e27.bin just now and the AP mode is exposed and I can connect and control [this bulb](https://www.banggood.com/ARILUX-E27-11W-RGBW-Dimmable-Smart-WIFI-LED-Light-Bulb-Works-with-Amazon-Alexa-Echo-AC85-265V-p-1175196.html?cur_warehouse=CN). FYI, the packaging (generic) says it's 7W not 11 as stated in the listing.

stelgenhof, 2018-05-08:
Some lightbulbs actually use different LED driver chips to control the various LED's on the PCB. The AiLight firmware only works with the MY9291 LED driver, so any bulb using that driver will work.

I believe the AriLux is using a different driver. Other firmwares like Espurna or Tasmota are compatible with a variety of lightbulbs/drivers so that explains why they work :)

xbmcnut, 2018-05-08:
I can confirm the Arilux bulb I have has a MY9291 LED driver. I'll reflash again this evening and see if it works 2nd time around.

stelgenhof, 2018-05-08:
You're correct. The Arilux E27 is indeed using the MY9291 driver. Was mixing it with another bulb I had in mind. Let me know if you managed to flash it (or not).

badgerhome, 2018-05-08:
Looking forward to seeing If you get this to work. Looks like they are 11w bulbs so would be interesting to see how brightness compares to the others.

xbmcnut, 2018-05-08:
Tried re-flashing this evening and again no AP mode. Seems device is constantly rebooting. PuTTy on COM8 @ 115200 below.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v02f0c112
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v02f0c112
~ld
```

stelgenhof, 2018-05-08:
Ah yes. It's same as the OP mentioned. Are you just flashing the binary release file, or compiling yourself?

xbmcnut, 2018-05-08:
Binary release. Very rusty on Platformio.

xbmcnut, 2018-05-08:
Binary on both instances. Ailight and Espurna.

xbmcnut, 2018-05-08:
Should it work if I build with Platformio?

xbmcnut, 2018-05-08:
SUCCESS! Compiled with Platformio and it is working a treat now. What's the difference between that and your compiled binaries?

xbmcnut, 2018-05-08:
Ah yes, see this now. Build is running the version below which is working a treat. Thanks for all your hard work and advice.

[image]

stelgenhof, 2018-05-08:
Super! Thanks for the testing and patience. My lightbulbs are working fine as well on this build. Guess I will prepare a release soon :)

xbmcnut, 2018-05-09:
For those wishing to flash these four channel bulbs via MQTT, here is the json.

`{"flash":"10", "brightness":255,"white_value":0,"color_temp":500, "color": {"r": 255, "g": 0, "b": 0}}`

xbmcnut, 2018-05-09:
I'm going to use the above when any of my smoke alarms goes off.

stevesoltys, 2018-05-17:
Just here to confirm that I was able to flash the latest development build of AiLight on this bulb: https://www.amazon.com/gp/product/B077VL5JXR/

It appears to be the same as the Arilux E27 ([packaging and all](http://tinkerman.cat/wp-content/uploads/2017/11/20171109_142859s.jpg)) but without the branding on the side.

Thanks for the fix!

stelgenhof, 2019-04-08:
Included in the just released version 0.6.0.

closed 2019-04-08 by stelgenhof