stelgenhof/AiLight #28: AiLight not working with provided binaries.

stelgenhof, 2018-02-04:
(Reported by Kai via separate conversation)

I erased the flash and written the ailight0.50.bin using esptool now in order to get more output. Everything seems fine but I still can't see an AP. I flashed at region 0x000000. Is this maybe the problem because I override the ESP Bootloader and the ailight.bin does not come with a bootloader? Here are the logs, it would be super kind if you could give me a hint what might be the prolem. I also tried building it myself on Windows using PlatformIO, but it gets confused with nodejs gulp because of Windows \ instead of / in PATH. Anyway here are the logs from esptool:

```
C:\Python27\Scripts>esptool.py.exe --port com3 erase_flash
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 10.9s
Hard resetting...

C:\Python27\Scripts>esptool.py.exe --port com3 write_flash 0x000000 ai.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0340
Compressed 360304 bytes to 259766...
Wrote 360304 bytes (259766 compressed) at 0x00000000 in 23.3 seconds (effective 123.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

Flashing espurna using the following command worked fine:

`esptool.exe -cd nodemcu -cb 116200 -cp COM3 -ca 0x000000 -cf espurna.bin`

however the same command with ai-0.50.bin did work (did not start an WiFi AP to connect to)

So just wanted you to know maybe there is a bug or you could figure out why it did not work.

stelgenhof, 2018-02-04:
Sorry to hear it didn’t work yet. Usually I don’t flash binaries this way, so maybe indeed I’ve overlooked something.

I've been trying to get some results but no luck so far. It also seems that the Arduino ESP8266 v2.4.0 has some issues. People have been reporting that their code increased in size significantly.

stelgenhof, 2018-02-04:
Attached is a new binary, based on the latest changes/fixes

[ailight-0.5.1-dev.zip](https://github.com/stelgenhof/AiLight/files/1692638/ailight-0.5.1-dev.zip)

kaiomatico, 2018-02-07:
Hey, Kai here, I just tried the new dev build and it is working fine! However it only works using this command: esptool.exe -cd nodemcu -cb 115200 -cp COM3 -ca 0x000000 -cf ailight-0.5.1-dev.bin

which means the windows port of esptool.py is able to flash it, but the program somehow is not able to execute afterwards...anyways great to see that the new build is working fine!

kaiomatico, 2018-02-07:
I know this should be opened in a new issue, but I don't know if the problem is only in this dev firmware: the color temperature is not saved after a reboot. I like my lights "warm" 👍

stelgenhof, 2018-02-08:
@kaiomatico Great to hear the flashing worked! After flashing, it is often need to restart the device. Usually, disconnecting the USB cable and then putting your bulb into a light socket is sufficient :)

stelgenhof, 2018-02-08:
@kaiomatico Could you give a bit more information regarding your HA setup? I've tested it myself and my light appears in HA perfectly. Perhaps your settings are a bit different.

kaiomatico, 2018-02-08:
Sure, thanks for offering help. I was using hassio with the addon mosqito-mqtt. My config looked like this:
```
mqtt:
  broker: core-mosquitto
  port: 1883
  username: ha
  password: ha
  discovery: true
  discovery_prefix: homeassistant
```
This worked fine and instantly using espurna with Home Assistant discovery, but not on AiLight. As you have seen in my logs, the AiLight really connected in the mqtt log. After much time testing around I tried switching to Home Assistants built-in embedded mqtt broker (also because no matter how many restarts I did, the old espurna lights were not deleted from Home Assistant). So my new config looks like this:

`mqtt:
  embedded:
  discovery: true
  discovery_prefix: homeassistant
`
In the logs I can see that the light shows up every second
`2018-02-08 08:47:43 INFO (MainThread) [hbmqtt.broker] Listener 'default': 1950 connections acquired
2018-02-08 08:47:43 INFO (MainThread) [hbmqtt.broker] Connection from 192.168.3.40:24576 on listener 'default'
2018-02-08 08:47:53 INFO (MainThread) [hbmqtt.broker] Listener 'default': 1951 connections acquired
2018-02-08 08:47:53 INFO (MainThread) [hbmqtt.broker] Connection from 192.168.3.40:27393 on listener 'default'
`
I even tried not using the discovery and configuring the light myself in the configuration.yaml, it shows up but is always set to off and I only have a brightness slider in Home Assistant which does nothing. Here is that part from the configuration.yaml
`2018-02-08 08:47:43 INFO (MainThread) [hbmqtt.broker] Listener 'default': 1950 connections acquired
2018-02-08 08:47:43 INFO (MainThread) [hbmqtt.broker] Connection from 192.168.3.40:24576 on listener 'default'
2018-02-08 08:47:53 INFO (MainThread) [hbmqtt.broker] Listener 'default': 1951 connections acquired
2018-02-08 08:47:53 INFO (MainThread) [hbmqtt.broker] Connection from 192.168.3.40:27393 on listener 'default'
`
One last question: You did not respond to the color temperature problem not being saved after a restart, can you reproduce it with this firmware? This is kind of my most "important" problem right now, because I often use the light as dumb light by switching it on and off using the light switch, hehe  :-D

Thank you for your time and help!

stelgenhof, 2018-02-08:
Can't see anything wrong from a first look. Before checking the HA configuration, can you check if you can change the lights settings (e.g. color, brightness, etc.) using the Web UI? And also can you check if the REST API and MQTT interface work?

stelgenhof, 2018-02-09:
I've done some more tests uploading the latest binaries and my settings (including color temperature) stay as they were after a reset (or unplugging).

As for the MQTT Discovery, have you enabled it through the UI like this?

[image]

I did notice that HA does recognize the light, but only the switch part is working :( Perhaps something has changed in HA the way MQTT discovery works. Let me investigate further.

kaiomatico, 2018-02-09:
Ok thanks for the feedback, I actually did not enable the lights like that (because I entered the MQTT IP Address and Port number)
I also found out that the color temperature actually stays, but is reported wrong in the web UI (the red, green, blue sliders are in the correct position for the color temperature, but the color temperature slider is set back to it's default value of 244).

But I finally managed to get the lights working properly even with RGB control etc. On hassio I had to install mosqiot mqtt broker again, leave the configuration default (so no username/password, anonymous read/write access) and then my configuration.yaml looks like this:
````
mqtt:
  broker: core-mosquitto

light:
  platform: mqtt_json
  name: 'AiLight-Schlafzimmer'
  state_topic: 'AiLight-Schlafzimmer'
  command_topic: 'AiLight-Schlafzimmer/set'
  color_temp: true
  brightness: true
  rgb: true
  white_value: true
  effect: false
  optimistic: false
````
And these are the AiLight settings (note: I needed to restart the light for them to apply!)
[image]

Thank you for helping me out and thank you for the awesome firmware! By the way a little feature request: could you maybe enable an option to let the light always start in the "ON" state? This would help if it was previously turned off by home assistant, then hard shut off by a light switch, then shut on by a light switch -> and then it just stays dark but you would normally like it to turn on using the light switch in the ON position

Thank you!

stelgenhof, 2018-02-09:
Good to hear you got it working! I indeed have seen that the color temperature value doesn't keep its last set value. I'll investigate that. 

Although I don't use the MQTT discovery feature myself, I do see that the light isn't properly loaded/configured in HA. 

Yes, having the option to keep the light 'ON' (or 'OFF') is a feature already a few others have requested. That is my next add-on :)

Please don't hesitate to log an issue if you have found something that isn't working properly or if you have a new feature you'd like to see added. :)

stelgenhof, 2018-02-11:
@kaiomatico I am closing this issue as I have created new issues for the color temperature and HA discovery ones separately.

closed 2018-02-11 by stelgenhof

nomadrick, 2018-08-28:
Would someone be kind enough to give me the WiFi password for the "ailight-0.5.1-dev.bin"?

Thanks.

stelgenhof, 2018-08-28:
@nomadrick If you're talking about *your* WiFi password, you should check your WiFi router/modem settings. :) However, assuming you would like to know the AiLight admin password, it is "hinotori" (without the quotes).

Cheers! Sacha

moishap, 2018-12-04:
I flashed the f/w and my bulb is booting up in AP mode with AiLight-XXXX SSID. What is the pw for that AP ? Thanks !

stelgenhof, 2018-12-04:
@moishap The default password for both the HTML UI admin and AP mode is "hinotori".