stelgenhof/AiLight #31: Compatible bulbs?

zmrow, 2018-02-17:
Hey there - I've done a bit of searching using the link for compatible 9W bulbs on your wiki and it seems these bulbs have disapeared!  Do you still have a good source for 9W bulbs that work for this firmware?  Please and thank you!

BTW, wonderful job on this firmware - it's truly fantastic.  :) 👍

stelgenhof, 2018-02-18:
I haven't shopped recently so am not sure what is a good source these days. It is possible indeed that some stores don't offer the bulbs any longer or only the 7W version. I have found also that the 9W is hard to come by.

zmrow, 2018-02-24:
FYI - I've received this bulb:  https://www.amazon.com/gp/product/B079BCYXSW/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1

It appears to have the correct driver and I've attempted to flash it.  Platformio completes the flash successfully, however nothing comes up on the serial monitor after a reset (unplug and plug of the usb cable).  Does the bulb need to be connected to mains to get output on the serial monitor?

stelgenhof, 2018-02-24:
If you flashed with the 'dev' settings then you should get output. When using 'prod' then all output is disabled.

zmrow, 2018-02-24:
Ah - you're absolutely right.  I realized that later.

Tried again to flash 'dev' and got the below:
`Linking .pioenvs/dev/firmware.elf
Checking program size
text       data     bss     dec     hex filename
428803    10096   32168  471067   7301b .pioenvs/dev/firmware.elf
Error: The program size (438899 bytes) is greater than maximum allowed (434160 bytes)
*** [.pioenvs/dev/firmware.elf] Explicit exit, status 1
========================== [ERROR] Took 3.40 seconds ==========================
 
================================== [SUMMARY] ==================================
Environment dev-ota     [SKIP]
Environment prod        [SKIP]
Environment prod-ota    [SKIP]
Environment dev         [ERROR]
========================== [ERROR] Took 3.40 seconds ==========================`

Should I open a different issue for this?

Any tuning I can do to reduce the firmware size?

stelgenhof, 2018-02-25:
Can you tell what version of the firmware you are using/flashing? There have been many people noticing a size increase of their program size after upgrading to ESP8266/Arduino v2.4.0.

zmrow, 2018-02-25:
This appears to be a weird PlatformIO/Arduino issue as you referenced here:  https://github.com/stelgenhof/AiLight/issues/27

I was able to flash "dev" using the ```firmware.bin``` and ```esptool```.  The light is working fantastically now!

stelgenhof, 2018-02-25:
Yes, I am not yet convinced if everything works fine with v2.4.0. At the moment I am kind of waiting until the storm has settled and hopefully an update is released :)

Happy to hear that you finally got it working, and sorry for the troubles!

zmrow, 2018-02-26:
Appreciate the help!  This firmware is fantastic!

closed 2018-02-26 by zmrow

stelgenhof, 2018-02-26:
@zmrow Thanks! Some more updates in the works... :)