stelgenhof/AiLight #32: Errors during compile 'WIFI_SSID' was not declared in this scope

rust84, 2018-03-07:
Encountering an error when trying to compile. Configured the default values for wifi and mqtt in src/config.h. The compiler is complaining about variables not being declared.

I noticed that if I comment out the wifi configuration (default) then the build finishes ok. 

Am using platformio installed with pip. Tested on Windows and Linux and it is the same. See my config below.

`******************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
******************************************************************************************************************************************************

[Wed Mar  7 13:20:34 2018] Processing dev (extra_scripts: build.py; build_flags: -DDEBUG -g -Wall -Wl,-Tesp8266.flash.1m128.ld; lib_deps: ArduinoJson, AsyncMqttClient, xoseperez/my9291, ESPAsyncTCP, ESP Async WebServer; platform: espressif8266; board: esp8285; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=64
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=346
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ ^1.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826 @ ^1.0.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing my9291
git version 2.7.4
Cloning into '/home/russell/repos/AiLight/.piolibdeps/_tmp_installing-69RTfQ-package'...
remote: Counting objects: 20, done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 20 (delta 0), reused 10 (delta 0), pack-reused 0
Unpacking objects: 100% (20/20), done.
Checking connectivity... done.
LibraryManager: Installing id=306
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PlatformManager: Installing espressif8266
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
espressif8266 @ 1.6.0 has been successfully installed!
The platform 'espressif8266' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
PackageManager: Installing tool-esptool @ ~1.409.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
PackageManager: Installing framework-arduinoespressif8266 @ ~1.20400.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
CorePackageManager: Installing tool-scons @ ~3.20501.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
tool-scons @ 3.20501.2 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
Converting main.ino
Collected 32 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <ArduinoJson> v5.13.1
|-- <AsyncMqttClient> v0.8.2
|   |-- <ESPAsyncTCP> v1.1.3
|   |   |-- <ESP8266WiFi> v1.0
|-- <my9291> v2.0.0 #fd9271c
|-- <ESPAsyncTCP> v1.1.3
|   |-- <ESP8266WiFi> v1.0
[... 355 lines cut ...]
define MQTT_PORT 1883
define MQTT_SERVER "192.168.1.81"
#define MQTT_USER ""
#define MQTT_PASSWORD ""
#define MQTT_RECONNECT_TIME 10000
#define MQTT_QOS_LEVEL 0
#define MQTT_RETAIN false
#define MQTT_KEEPALIVE 30

#define MQTT_PAYLOAD_ON "ON"
#define MQTT_PAYLOAD_OFF "OFF"

#define MQTT_STATUS_ONLINE "online"
#define MQTT_STATUS_OFFLINE "offline"

#define MQTT_HOMEASSISTANT_DISCOVERY_ENABLED false
#define MQTT_HOMEASSISTANT_DISCOVERY_PREFIX "homeassistant"

/**
 * HTTP
 * ---------------------------
 * Use the below variables to set the default HTTP settings of your Ai-Thinker
 * RGBW Light. These will be used as the factory defaults of your device.
 */
#define REST_API_ENABLED false`

stelgenhof, 2018-03-07:
I think you made a typ-o of some sorts when editing the config.h file. Most likely you didn't close a string (using quotes or double quotes) properly in the config.h file. If that occurs, then basically the syntax is garbled which could explain these errors.

Can you please check if your config.h file is correctly edited?

rust84, 2018-03-07:
It was my fault for removing the # thinking it was already commented out. Been a long time since I did anything with c! Thanks for responding anyway.

closed 2018-03-07 by rust84