stelgenhof/AiLight #35: Feature Request: MQTT availability_topic

SurfingSpider, 2018-03-26:
Can I ask if we can get the availability_topic working. So if the bulb is off it shows as unavailable in home assistant.

[https://www.home-assistant.io/components/light.mqtt/#availability_topic](https://www.home-assistant.io/components/light.mqtt/#availability_topic)

I have it working on my tasmota sonoff switches and it is very handy to see if one of them goes offline

SurfingSpider, 2018-04-06:
I see on the front page that you have enabled MQTT Last Will and Testament maybe this is all i need. just put it in the availability topic. I cant find any info on the wiki on how to implement it.

When I setup a device in home assistant (mosquitto Mqtt) which has **tasmota firmware** on it my configuration.h looks like this.....

```
light:
  - platform: mqtt
    name: "Fireplace" 
    state_topic: "stat/sonoff7/POWER" 
    command_topic: "cmnd/sonoff7/power" 
    availability_topic: "tele/sonoff7/LWT"    
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true
```

I assume that the availability_topic would be the Status Topic in the AiLight UI. and the Payload_on, Payload_off, Payload_availible, Payload_Not_available would be the responses AiLight gives over MQTT

Do you think I could be close to the mark?

mjoshd, 2018-04-07:
I didn't have to do anything special for the LWT online/offline messages to
work. About 30-60 seconds after the bulbs lose power, the MQTT broker will
send the "offline" message and moments after power is restored the "online"
message is sent.

Here is what I have in my config-

light:
- platform: mqtt_json
name: My Lamp
state_topic: "hassio/ailight/my_lamp"
command_topic: "hassio/ailight/my_lamp/set"
color_temp: true
brightness: true
rgb: true
white_value: true
effect: false
optimistic: false

mjoshd, 2018-04-07:
I didn't have to do anything special for the LWT online/offline messages to
work. About 30-60 seconds after the bulbs lose power, the MQTT broker will
send the "offline" message and moments after power is restored the "online"
message is sent.

Here is what I have in my config-

```
light:
  - platform: mqtt_json
    name: My Lamp
    state_topic: "hassio/ailight/my_lamp"
    command_topic: "hassio/ailight/my_lamp/set"
    color_temp: true
    brightness: true
    rgb: true
    white_value: true
    effect: false
    optimistic: false
```

stelgenhof, 2018-04-07:
As @mjoshd you don't have to do anything for MQTT Last Will and Testament to work. You can read more about it here: https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament

I can add the availability topic as well, although I am not 100% sure how this and MQTT LWT relate to each other. They seem to serve the same purpose. Guess I need to do some more reading... :)

stelgenhof, 2018-04-07:
@SurfingSpider I have added the availability topic (as part of the auto discovery feature of Home Assistant). It uses the same MQTT topic as used for the Last Will and testament.

The payloads for available/unavailable are the defaults as used in Home Assistant ("online" and "offline").

The changes are done in the default ('develop') branch and will be released once considered ok/stable.

Cheers! Sacha

hulkhaugen, 2018-04-12:
umm, i'm trying to add this manually myself. MQTT is still pretty new to me, but i've tried a lot. I cannot get it to work with /set, /status or just blank. Currently it looks like this:

```
- platform: mqtt_json
  name: 'WiFi Smart Bulb 1'
  command_topic: "WiFi_Smart_Bulb_1-43FC47/set"
  state_topic: "WiFi_Smart_Bulb_1-43FC47"
  availability_topic: "WiFi_Smart_Bulb_1-43FC47"
  payload_available: "Online"
  payload_not_available: "Offline"
  qos: 1
  color_temp: true
  brightness: true
  rgb: true
  white_value: true
  effect: false
  optimistic: false
  retain: true
```

I've also tried to listen to the different topics in terminal, but i can't find any online/offline status in the 3 different topics. So, is there a 4th hidden topic i should listen to?

BTW, i'm also unable to change the name of the devices and their topics, they just revert back to what they were set as during flash...

stelgenhof, 2018-04-12:
@hulkhaugen As I just implemented this, I haven't yet updated the documentation. Apologies for that.

As for the "availability_topic", it is the same as the status topic, so will need to look like this:
"<YOUR DEVICE ID>/status"

Also your `payload_available` and  `payload_unavailable` are not correct. These values are the defaults as used in Home Assistant ("online" and "offline"). Unless you changed these values in your config.h file: 

#define MQTT_STATUS_ONLINE "online"
#define MQTT_STATUS_OFFLINE "offline"

hulkhaugen, 2018-04-13:
Thanks for your quick reply. Is there something i'm missing here? Unless you refer to the payloads being case sensitive in HA, i can't see the issue. You also writing `payload_unavailable` instead of `payload_not_available`, but i guess that's on you. However i tried all them combinations, but it just shows as unavailable, even when it's on (before restart HA). It currently looks like this:

```
- platform: mqtt_json
  name: 'WiFi Smart Bulb 1'
  command_topic: "WiFi_Smart_Bulb_1-43FC47/set"
  state_topic: "WiFi_Smart_Bulb_1-43FC47"
  availability_topic: "WiFi_Smart_Bulb_1-43FC47/status"  
  qos: 1
  payload_available: "online"
  payload_unavailable: "offline"
  color_temp: true
  brightness: true
  rgb: true
  white_value: true
  effect: false
  optimistic: false
  retain: true
```

I'm using 0.5.0-dev as of 24 Nov 2017.

stelgenhof, 2018-04-13:
Sorry, my mistake. It indeed should be `payload_not_available` (Typo).

The firmware implements the mqtt_json light of Home Assistant. You can find all information here: https://www.home-assistant.io/components/light.mqtt_json/

The availability_topic feature has only been recently implemented in HA (not sure exactly since which version), however I checked the source code you are using (0.50) and it was definitely not there :)

Guess you need to do an upgrade of HA.

hulkhaugen, 2018-04-13:
No, i'm using HA 0.66.1. It's the AiLight firmware that's v 0.5.0-dev.

stelgenhof, 2018-04-13:
Ah ok! 

Anyway, AiLight 0.5.0-dev doesn't have this feature :) You need to compile the firmware yourself  (from the develop branch) as it isn't officially released yet.

stelgenhof, 2019-04-08:
Included in the just released version 0.6.0.

closed 2019-04-08 by stelgenhof

xbmcnut, 2019-11-24:
I would like to implement this too but have HA restart the bulb if it's offline for more than _x_. Is restart via the API possible?

stelgenhof, 2019-11-24:
@xbmcnut Not sure what feature you are referring to. If it is the availability_topic, that is already there. As for the restart via the API (or MQTT), that is currently not possible.

xbmcnut, 2019-11-24:
Thanks, appreciate the quick reply. Is restart via HTML URL possible?

stelgenhof, 2019-11-24:
Sorry, via the URL neither. The only way is via the Web UI which actually uses WebSockets. At the moment WebSockets would be the only way.

If you feel you'd like to have other options for restarting the device (via REST API, URL, MQTT), I'd suggest to create a new issues (feature request).

Cheers! Sacha

xbmcnut, 2019-11-24:
Thanks, will do.