stelgenhof/AiLight #36: [Request] Rainbow effect

hulkhaugen, 2018-04-12:
I really like this firmware, but i do however miss the lack of effects. I think the rainbow effect is probably the most wanted one, at least for me. If you could look at a way to integrate this, that would be awesome.

https://github.com/xoseperez/my92xx/blob/master/examples/esp/esp8285_rainbow.cpp

It's all greek to me, so i don't know if the transition times in the effect are adjustable or what's what. If that could be done, it would be the icing on the cake. Cheers and thanks for awesome work on this firmware.

stelgenhof, 2018-04-12:
Thanks for the feature request. Effects are on the Todo list, however not many have requested it :) Personally I don't have a need for effects (yet), but can definitely look at implementing it.

hulkhaugen, 2018-04-13:
Awesome, thank you. Yes the rainbow seems "useful" to me, most others makes more sense for a addressable LED strip. Maybe a blink/strobe effect are interesting to someone, or "police lights" alternating between two colors. Personally, i think that would be annoying as a practical use, but maybe fun to show off :D

stelgenhof, 2018-04-13:
FYI: The flash memory in these lights are pretty limited (usually only 1MB) and having many effects implemented would consume a lot of memory. So the number of effects will be pretty small.

hulkhaugen, 2018-04-13:
Yes, I understand that. Just the rainbow and I will be a happy puppy :)

closed 2020-03-05 by github-actions[bot]