stelgenhof/AiLight #39: Freezes for a couple of mins when firing quick MQTT

badgerhome, 2018-05-08:
Hi,

I'm testing 0.6.0-dev, I'm using it with this script: https://community.home-assistant.io/t/finally-a-cheap-wireless-switch-that-dims-xiaomi-switch-gen1/27574

Which basically fires a dim command in a loop until a button press is released:
```
service: light.turn_on
        entity_id: light.landing_light
        data_template:
          brightness: '{{states.light.landing_light.attributes.brightness - 10}}'
```
I've noticed that 0.6.0-dev freezes up for a couple of mins and then comes back, I can still access thr bulb via HTTP but can't issue any commands, I've downgraded the bulb to 0.5.1-dev and it seems to be working fine.

Sorry I can't provide any more details, I'll carry out doing some more testing. Let me know if there's anything else useful I can share.

Cheers!

badgerhome, 2018-05-08:
in fact I can't access via HTTP at all when it freezes but I am able to ping it.

stelgenhof, 2018-05-09:
I have not seen this myself. Need to use your script myself then to see if I can replicate the issue.
Also I checked the code changes between 0.5.1-dev and 0.6.0-dev and there isn't anything particular that could be the cause of this (for now).

stelgenhof, 2018-05-23:
@badgerhome I have been able to find some time testing your issue. 

Since I didn't have your setup with the switch, I used a small script to repeatedly fire a MQTT message to the light. Unfortunately I couldn't replicate the issue that it was freezing up after running the script. It continued to operate perfectly. REST API, MQTT and Web UI were accessible as normal.

I've also test it at different times of the day :) Just to make sure.

Not sure what I can do to replicate your issue.

stelgenhof, 2019-01-06:
@badgerhome Just wanted to follow up if you are still experiencing this issue. (Cleaning up old issues :) )

stelgenhof, 2019-02-13:
@badgerhome 
After doing some testing regarding the freezing of the firmware, I have made some changes to the platformio.ini configuration. Basically locked down the versions of the ESP8266 framework and a few dependent libraries.

Please adjust the following in your platformio.ini file:
`platform = espressif8266@1.5.0`

```
lib_deps:
ArduinoJson@5.13.4
AsyncMqttClient@0.8.2
```

My tests showed some good results. Can you please have try with these settings as well? (Checkout the latest repository changes). If this seems to be working, then we have a solid base.

Cheers! Sacha

stelgenhof, 2019-04-08:
Improvements included in the just released version 0.6.0.

closed 2019-04-08 by stelgenhof