stelgenhof/AiLight #40: HA MQTT auto discovery problem

nickrout, 2018-05-22:
My light running AiLight 0.51-dev was working perfectly so of course like an idiot I changed some things.

I got a new wifi AP so I changed the essid and password. That has worked out fine.

At the same time I thought I'd change the name of the bulb to something more descriptive, so I changed the following in the gui:
hostname: hallwaylight
state topic: hallwaylight
command topic: hallwaylight/set
status topic: hallwaylight/status
MQTT Discovery is on, Discovery prefix is homeassistant (both unchanged). 

However the bulb never publishes a discovery message. I am watching the entire MQTT namespace via mosquitto_sub, and nothing in the homeassistant/ space, only the following:

mosquitto_sub -v -h xxxx -u xxxxx -P xxxxxxx -t '#' -v
hallwaylight {"state":"ON","brightness":255,"white_value":255,"color_temp":244,"color":{"r":0,"g":0,"b":0},"gamma":true}
hallwaylight/status online

Any ideas?

stelgenhof, 2018-05-22:
Not sure. Did you recompile/reflash the bulb?
Perhaps a restart of HASS might help.

nickrout, 2018-05-22:
No I hadn't recompiled or reflashed, I simply made the changes over the bulb's web server.

I just tried:

- turned off mqtt discovery, saved, and restarted bulb
- turned on mqtt discovery, saved and restarted bulb 

It is now working as expected, with discovery all good. I am not sure why it wasn't working before, perhaps a browser issue (firefox 59.0.2). I did think I'd tried that sequence previously, but perhaps I'd not done it properly.

stelgenhof, 2018-05-22:
Ah! I remember now that the firmware stores whether the light already has been discovered by HASS or not. This to avoid repeatedly broadcasting a discovery call. I need to have a look at the code.

Maybe the code is too strict, i.e. should allow discovery again if something has changed (i.e. name, WiFi, etc.)

nickrout, 2018-05-22:
Damn I do remember reading that a while ago but I had forgotten. 

Probably no need to change it on wifi change, as you might still be on the same mqtt server, ie you are still in the same home. But a change of hostname or mqtt server details should probably allow a re-broadcast. Up to you of course.

stelgenhof, 2018-05-22:
I had a look at the code. MQTT Auto discovery is triggered upon connect to the MQTT broker and if the device hasn't been auto discovered yet.

The most critical pieces of the auto discovery in this regards are:
- hostname
- state_topic
- command_topic

So I will make a change that if any of these 3 settings are changed, the device hasn't been discovered any longer.

stelgenhof, 2018-05-23:
@nickrout I have made some minor code changes (develop branch) that will trigger a re-discovery if also the state topic, command topic and status topic are changed.

As a matter of fact, just changing the hostname already triggered the device to re-register itself in Home Assistant. So, your steps to restart the device are actually not needed. See my screenshot below:

[image]

My device was first named "AiLight-0F232C" and then I changed it's hostname to "office". Pressing the "Save" button resulted in the device to appear in Home Assistant as "office". (The 'old' version is still present, but becomes 'unavailable' after some time).

That being said, I believe it was still working as designed. Not sure why in your case the auto discovery message wasn't being broadcasted initially.

stelgenhof, 2018-05-26:
I know you're using an older version of AiLight, however would you mind testing the latest version (develop branch) to see if the auto discovery is still an issue for you?

As said, on my side it all looks fine, but always good to get a second opinion :)

Cheers! Sacha

stelgenhof, 2019-01-06:
@nickrout Can you let me know if you are still having an issue with the auto discovery?

closed 2019-01-09 by stelgenhof