stelgenhof/AiLight #42: behaviour on wifi disconnect

lxz81, 2018-05-28:
Hi,

If my wifi network can not be reached, my bulbs fall back to access point mode. 
Is it possible to let the bulb retry connecting to the wifi network (and mqtt server) every now and then?

stelgenhof, 2018-05-28:
Can you provide a bit more information? Which version of the firmware are you using? Also - if you happen to know - what version of the ESP8266 framework are you using?

Cheers! Sacha

stelgenhof, 2018-05-29:
@lxz81 Thanks for the info! Let me have a look at the code of 0.4.1 first.

packetinspector, 2018-05-30:
I have.  I think my wifi takes longer to "boot" than others.  Power goes out, bulbs never reconnect.  I have to either power cycle or login and reset them.

stelgenhof, 2018-05-30:
Ok. I will do some testing on my side and see if I can make some improvements :)

nphil, 2018-06-26:
having the same issue, if WiFi goes out, the bulbs don't automatically reconnect and go into AP mode.

badgerhome, 2018-10-21:
Is this fixed in the latest release? Planning on updating mine this evening. I've had some building work done at home recently and my AP has been up and down a lot with the bulbs going into AP mode as the WiFi takes longer to start up on my router than the boot time of the bulbs so they keep going into AP mode. Thanks!

stelgenhof, 2018-10-21:
@badgerhome I'm pretty sure it is not fixed yet. Last time I worked on the firmware was about 2 months ago and fixed only some other issues. 

Perhaps this week I may have some time to investigate. Sorry about the delay.

Cheers! Sacha

mjoshd, 2019-02-21:
Really hoping this makes it into the next update. I have multiple bulbs that I have to manually power cycle any time my AP becomes temporarily unavailable from updating the firmware, power outage, etc; the bulbs go into AP mode and do not automatically reconnect to WiFi when it becomes available again.

stelgenhof, 2019-02-21:
@mjoshd At the moment there is no solution yet. I am aware of the issue, however haven't had the time to investigate.

brahmafear, 2019-02-21:
Brainstorming a temporary fix :

What about a setting that says a bulb in AP mode should reboot every hour ( or whatever interval ).  That would seem to go be enough time to initially program bulb and would cause this glitch to fix itself at most an hour after WiFi restored. 

Not a perfect solution, but might help.

packetinspector, 2019-02-21:
Another possibility:

While in AP mode and before the captive portal is hit for the first time, every X seconds do an AP scan.  If the saved wifi network is present, reboot or reinit.

stelgenhof, 2019-02-21:
Good suggestions. Let me do a little bit of investigating this weekend. I need to setup  a secondary WiFi network first so I can simulate this properly :)

stelgenhof, 2019-02-23:
@lxz81 @packetinspector @badgerhome @nphil 
After doing some more research, I've made a small change that may alleviate the issues when your WiFi connection drops. 

A new compiler directive was added (WIFI_RECONNECT_TIMEOUT) that you can use to set how long the device needs to try to connecting to your WiFi Access Point. Previously, this time period was not configurable and hard coded to 10 seconds. Once this time has been reached, the device will go the Soft AP mode (assuming a WiFi connection is not possible / not configured).

By default, the value is set to 60 seconds (1 minute). A fallback of 10 seconds is set in case you haven't added the new directive in your `config.h` file yet.

Please note that this change will not try and reconnect when in AP mode. For that, additional coding is required, which should be done carefully because the memory capacity is limited. With this solution, you can set the timeout period to a time that fits your situation best, which I think is a good alternative.

Hopefully this works for everybody and ask you to give it a try :)

Cheers! Sacha

stelgenhof, 2019-04-08:
Included the latest changes in the just released version 0.6.0. Please do not hesitate to create a new issue if you feel the issue hasn't been resolved yet.

closed 2019-04-08 by stelgenhof