stelgenhof/AiLight #43: problem with OTA update

CatTurbo, 2018-08-30:
Hello
First thank you very much for the development of the AiLight. It works great on my bulb and thanks to it I could develop an alarm clock which is controlled over a web application. Thanks. 
The upload over serial connection works without problem, but unfortanetely the OTA update doesn't work at all. 
When I'm trying to do an ota update with visual studio code (with  pio run --target upload -e  prod-ota --upload-port 192.168.1.198), I'm getting the following error while uploading:
`21:02:14 [DEBUG]: Options: {'timeout': 10, 'esp_ip': '192.168.1.198', 'host_port': 45376, 'image': '.pioenvs\\prod-ota\\firmware.bin', 'host_ip': '0.0.0.0', 'auth': 'hinotori --port=8266', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
21:02:14 [INFO]: Starting on 0.0.0.0:45376
21:02:14 [INFO]: Upload size: 397936
Sending invitation to 192.168.1.198
Authenticating...FAIL
21:02:14 [ERROR]: Authentication Failed`

I didn't change the standard platformeio.ini and config.h so the port should be 8266 and the password hinotori as described in the error message. The webinterface of the bulb listen on 192.168.1.198, so the ip is correct. 

The following things I've tried:

- The upload over ota doesn't work for me in the 0.5 version and the actual 0.6 development version
- the firewall of windows wasn't on while testing
- In the platformeio.ini file I've tried to add an = after port in the upload flags, but it didn't work either
` upload_flags = --auth=${common.ota_password} --port=${common.ota_port}`
-I've tried the other ota build (dev-ota) too, but unfortanetely without result. 
-I've tried to add the --port in upload_flags on a new line after --auth=, but then the compilation didn't work. 

So now I don't know what I can try anymore, I'm very glad for any help. 
thanks you very much. 
Greetings 
Cat Turbo

Here is my platformeio.ini file for reference (shouldn't be any changes to original one):
`; Ai-Thinker RGBW Light Firmware - Project Configuration File
;
; This file is part of the Ai-Thinker RGBW Light Firmware.
; For the full copyright and license information, please view the LICENSE
; file that was distributed with this source code.
;
; Created by Sacha Telgenhof <me at sachatelgenhof dot com>
; (https://www.sachatelgenhof.nl)
; Copyright (c) 2016 - 2018 Sacha Telgenhof

[platformio]
env_default = dev

[common]
platform = espressif8266
framework = arduino
upload_speed = 115200
upload_port = "<YOUR_DEVICE_ID>.local"
ota_password = hinotori
ota_port = 8266
flag_flash_size = -Wl,-Tesp8266.flash.1m128.ld
flag_debug = -DDEBUG -g -Wall
lib_deps =
  ArduinoJson
  AsyncMqttClient
  xoseperez/my9291
  ESPAsyncTCP
  ESP Async WebServer

# Development/Debug environment
[env:dev]
platform = ${common.platform}
board = esp8285
framework = ${common.framework}
build_flags = ${common.flag_debug} ${common.flag_flash_size}
lib_deps = ${common.lib_deps}
extra_scripts = build.py

# Development/Debug environment for OTA Updates
[env:dev-ota]
platform = ${common.platform}
board = esp8285
framework = ${common.framework}
build_flags = ${common.flag_debug} ${common.flag_flash_size}
lib_deps = ${common.lib_deps}
extra_scripts = build.py
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
upload_flags = --auth=${common.ota_password} --port ${common.ota_port}

# Production optimized environment
[env:prod]
platform = ${common.platform}
board = esp8285
framework = ${common.framework}
build_flags = -Os ${common.flag_flash_size}
lib_deps = ${common.lib_deps}
extra_scripts = build.py

# Production optimized environment for OTA Updates
[env:prod-ota]
platform = ${common.platform}
board = esp8285
framework = ${common.framework}
build_flags = -Os ${common.flag_flash_size}
lib_deps = ${common.lib_deps}
extra_scripts = build.py
upload_speed = ${common.upload_speed}
upload_port = ${common.upload_port}
upload_flags = --auth=${common.ota_password} --port ${common.ota_port}`

and here is the config.h file:
`/**
 * Ai-Thinker RGBW Light Firmware - Configuration
 *
 * This file is part of the Ai-Thinker RGBW Light Firmware.
 * For the full copyright and license information, please view the LICENSE
 * file that was distributed with this source code.
 *
 * Created by Sacha Telgenhof <me at sachatelgenhof dot com>
 * (https://www.sachatelgenhof.nl)
 * Copyright (c) 2016 - 2018 Sacha Telgenhof
 */

/**
 * Light
 * ---------------------------
 * Use the below variables to set the default behaviour of your Ai-Thinker RGBW
 * Light. These will be used as the factory defaults of your device.
 */
#define LIGHT_STATE false
#define LIGHT_BRIGHTNESS 0
#define LIGHT_COLOR_TEMPERATURE 0
#define LIGHT_COLOR_RED 0
#define LIGHT_COLOR_GREEN 0
#define LIGHT_COLOR_BLUE 0
#define LIGHT_COLOR_WHITE 0

#define HOSTNAME "AiLight"
#define ADMIN_PASSWORD "hinotori"

#define POWERUP_MODE POWERUP_OFF

/**
 * OTA (Over The Air) Updates
 * ---------------------------
 */
#define OTA_PORT 8266

/**
 * WiFi
 * ---------------------------
 * Use the below variables to set the default WiFi settings of your Ai-Thinker
 * RGBW Light. These will be used as the factory defaults of your device. If no
 * SSID/PSK are provided, your Ai-Thinker RGBW light will start in AP mode.
 */
#define WIFI_SSID "my ssid is here"
#define WIFI_PSK "here is my wlan password"
#define WIFI_OUTPUT_POWER 18.0 // 20.5 is the maximum output power

/**
 * MQTT
 * ---------------------------
 * Use the below variables to set the default MQTT settings of your Ai-Thinker
 * RGBW Light. These will be used as the factory defaults of your device and
 * making the connection to your Home Assistant instance. Most of these settings
 * can also be changed in the UI environment.
 */
#define MQTT_PORT 1883
#define MQTT_SERVER ""
#define MQTT_USER ""
#define MQTT_PASSWORD ""
#define MQTT_RECONNECT_TIME 10000
#define MQTT_QOS_LEVEL 0
#define MQTT_RETAIN false
#define MQTT_KEEPALIVE 30

#define MQTT_PAYLOAD_ON "ON"
#define MQTT_PAYLOAD_OFF "OFF"

#define MQTT_STATUS_ONLINE "online"
#define MQTT_STATUS_OFFLINE "offline"

#define MQTT_HOMEASSISTANT_DISCOVERY_ENABLED false
#define MQTT_HOMEASSISTANT_DISCOVERY_PREFIX "homeassistant"

/**
 * HTTP
 * ---------------------------
 * Use the below variables to set the default HTTP settings of your Ai-Thinker
 * RGBW Light. These will be used as the factory defaults of your device.
 */
#define REST_API_ENABLED true
`

brahmafear, 2018-08-30:
Hi CatTurbo,

I had same issue. Despite the error saying “authentication failure”, the reason the the failure was insufficient flash space on the bulb. Ie, flash was not big enough to hold 2 copies of code (required for  ota updates). You can see this error message by running some of the command line programs manually. 

You might try doing the typical, 2 stage OTA update — do an ota upload of some very stripped down code that does nothing but wait for an ota update, then do an ota update of the newer AiLight code. 

Hope this helps.

CatTurbo, 2018-09-02:
Hello brahmafear
First thank you very much for your fast answer. I think you are completely right, my bulb has 1MB of flash, so space is IMHO here the problem. 
About the 2 stage update. I've tried it with the standard android OTA sketch for ESP8266

`#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "here is my wifi name";
const char* password = "here is my wifi password";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
   ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  ArduinoOTA.setPassword("hintori");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}`

The upload worked. In the next step I've tried to upload another code (the ailight code), but the PIO gave me a memory failure error. In the Arduino IDE the flash also didn't work too. 
 
Additionally I've got a huge problem to connect the bulb with a serial interface, cause the last time I desoldered the calbes, I've accidentally damaged the RX solder pad on the bulb (now it is broken and it don't have a connection to the ESP chip anymore) . So this time I had to make a connection directly to the ESP chip with a multimeter probe . It worked, but I don't think it will work a second time ;)
So do you think that the OTA update should work normally with the arduino code posted above or do I have to use another code for a two stage flash?
Thank you very much for a response

brahmafear, 2018-09-02:
Could you provide the exact error messages you received?

CatTurbo, 2018-09-03:
Good evening brahmafear

first thanks you again for your response. In the meantime I think I've found the bug. 
I've first reflashed the Ai-Light firmware on the bulb (over a serial connection). Then I've flashed the Arduino OTA code on the bulb with platforme IO (over OTA update). There it seems to be important that there isn't an upload port specified in the platformeio.ini file. 
```upload_flags = --auth="here your password" ```
and not 
```upload_flags = --auth="here your password --port here_your_port```
as specified in the platformeio.ini file of the ai-light project. 

As soon as I specify a port in the platform.ini file, I'm getting an authentification error. So in the next step I've flashed the ai-light code on the bulb, also with no upload port specified. It worked. 
So just for fun I've tried after to flash directly a new version of the Ai-light firmware on the bulb when there is an existing ai-light code on the bulb. It works (no memory, no space error), but here the important thing too is to not specify a upload port. Then it works without authentification error and I can it flash too without the two steps flash. 

After some more testing I've discovered that the port should be defined on a new line with two blanks as intending and a = after port:
 
```
upload_flags = --auth=${common.ota_password} 
   --port=${common.ota_port}
```

This is also described in the docs, here
http://docs.platformio.org/en/latest/platforms/espressif8266.html
and here  https://docs.platformio.org/en/latest/projectconf/section_env_upload.html 

With this modification it works without problem.

stelgenhof, 2018-09-04:
@CatTurbo @brahmafear Thanks for catching this issue. I can do a fix for you no problem. 

Yes, I know the last release was a year back :( Lame excuse, but busy with other projects at the same time, haha. 

There are some minor changes/fixes done since, so maybe I can do a minor release (v0.5.1) soon.

stelgenhof, 2018-09-04:
Solved in the develop branch. You may need to adjust your own platformio.ini file yourself. 

I will do a 0.5.1 release probably end of this week. 

Cheers! Sacha

stelgenhof, 2019-04-08:
Included in the just released version 0.6.0.

closed 2019-04-08 by stelgenhof