stelgenhof/AiLight #44: LOHAS bulb support

crisboarna, 2018-11-04:
Hi,
Although you have mentioned the LOHAS bulbs are incompatible due to them having an MY9231 controller, on a HA thread I noticed that somebody tested it out and that it could control the bulb, just that all the settings were resulting in different outputs than expected. 

Are there any plans to add support for this type of bulb as it reaches greater lux (1380 on [this one](https://www.amazon.com/LOHAS-Multicolor-Equivalent-Decorative-Compatible/dp/B078SQD6MQ/) with very good colored lux capability as well, especially compared to majority of Ai-Thinker bulbs reaching only ~600 lux, assuming you can find a 9W for sale.

This is also an "untapped" market segment as there is currently no HA integration route for LOHAS bulbs besides potentially using HA Cloud component which is not as safe or free, so I believe this has great potential and don't think I would be only one interested in it.

If you do not have any plans on supporting it, could you provide some pointers on how to go about attempting this integration myself as I am new but very interested in the subject. 

Thanks !

stelgenhof, 2018-11-04:
Hi @crisboarna,

Indeed I have no immediate plans to support other type bulbs however am keen to have a look. At the moment I don't own one yet so some testing would be not possible yet. Let me see if I have some time soon. Definitely looks interesting though!

Cheers! Sacha

darkfiberiru, 2019-01-12:
Pull request #50 Successful works if following is set in config.h

>#define MY92XX_TYPE MY92XX_MODEL_MY9231
>#define MY92XX_COUNT 2

See notes in pull request

stelgenhof, 2019-05-09:
@crisboarna The current development branch now supports LOHAS type bulbs (using the M92xx chip) thanks to @darkfiberiru. I don't own such a bulb, so can't do any testing. Therefor please compile the firmware yourself. 

There are some minor things still to be done (UI) but it should be useable already.

closed 2019-05-09 by stelgenhof

darkfiberiru, 2019-12-06:
Just for reference 
Someone figured out how to safely power lohas bulbs to flash them https://majenko.co.uk/blog/more-esp8266-hacking-lohas-led