stelgenhof/AiLight #45: Question. Qube light bulb

jimpastos, 2018-11-11:
I accidentally broke a Qube light bulb while moving some boxes and started to look into what I can salvage from its components.

Any chance AiLight can be used (or parts of it) to flash the bulb? The original is dead as Qube took down their servers. Would be good to have a project to revive them

From what i can tell it has an ESP8266EX and a bluetooth chipset. Not sure if it has any fancy LED controller or not

Im attaching some pics of the PCB. I can attach more if needed

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

stelgenhof, 2018-11-11:
Thanks @jimpastos for the photos. Indeed it has both an ESP8266EX and a CSR1010 Bluetooth chip for communications. It would be interesting to see what kind of LED driver it is using. From the photos I couldn't see one. Unless the LEDs are controlled in a different manner.

jimpastos, 2018-11-11:
I couldn’t see any specific driver. Seems it just controls them with N-channel Mosfets with a secondary board that connects to the LED pcb

stelgenhof, 2018-11-11:
I see. The AiLight is currently designed to work with a LED driver only, not with the ESP8266 driving any MOSFET's. It would be possible, however not (yet) designed to do so. 

Cheers! Sacha

jimpastos, 2018-11-11:
If I could figure out how to flash this with custom firmware, i can experiment with it. 
For now i can probably try to use the mosfet pcb and the led grid on a pi and attempt to drive with its GPIOs. 

Ideally running the ESP with wifi and bluetooth with something custom with be perfect

stelgenhof, 2018-11-13:
I think the flashing isn't the hardest part. You would just need to connect wires to the ESP8266 chip directly.

stelgenhof, 2019-01-05:
@jimpastos Just wanted to know if I can close this issue? Did you get around resolving your issues?

Cheers! Sacha

closed 2019-01-06 by stelgenhof