stelgenhof/AiLight #47: Backing up OEM firmware

nospam2k, 2018-11-27:
I have 4 of the same bulbs and I did the following to back up the OEM image.

sudo esptool.py --port /dev/cu.usbserial --baud 115200 read_flash 0x000000 0x100000 light1m.bin
sudo esptool.py --port /dev/cu.usbserial --baud 115200 read_flash 0x000000 0x400000 light4m.bin

I wasn't sure the size of the flash so I made both (esptool said it was 4m)

When I flashed that back to the bulb it just goes into a boot loop. Fortunately, I was able to flash AiLight and it works fine but I'd like to get a working backup of the OEM firmware.

stelgenhof, 2018-11-27:
Not sure how I can help with the OEM firmware in this case, however have you tried clearing the flash space before writing the OEM image?

nospam2k, 2018-11-27:
I understand. I just wondered if you have been able to successfully backup the oem firmware and if so what the procedure would be. I tried writing the image to an NodeMCU but the results are the same. Just boot loops.

stelgenhof, 2018-11-27:
I recall that I was able to restore the OEM firmware on one of my bulbs but that has been a while. 

It is possible that the OEM firmware is expecting a different boot loader or that it is using a different flash region. It's hard to tell as there may be so many parameters involved for the original firmware to work.

nospam2k, 2018-11-27:
I think you are right about the bootloader as. So I backed up 0x000000 0x400000. Does that not include the bootloader?

Here is the boot loop text:

load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 1.4.2(23fbe10) compiled @ Sep 22 2016 13:09:03
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

stelgenhof, 2018-11-28:
It likely includes the boot loader. However as said there may be various other factors involved. 

It is possible to define where the bootloader on the Flash memory resides. The Atmel MCU's for example are using fuses (i.e. variables) to control the operation of the firmware. One of them is the bootloader start address. If this particular fuse is not corresponding with the loaded firmware, the MCU will not start or renders errors.

Although the ESP8266 doesn't use something like fuses, but the manufacturer could have created a different / non-standard firmware. Perhaps to protect the firmware from being copied/hacked. 

Of course I am not sure about this, but it is a possibility. As such it is hard to tell how the OEM firmware was compiled and created, and thus making it hard to replicate a successful flashing of the original OEM firmware.

nospam2k, 2018-11-28:
Ok, thanks for the info. I have 3 other bulbs so I will see if I can't copy the firmware out of them. Not likely to use it because your firmware works great!

closed 2018-11-28 by nospam2k

nospam2k, 2018-12-20:
I just wanted to update this for others. I was able to backup and restore the firmware. I factory defaulted the bulb from within the app. I'm not sure if this is necessary or not because I didn't play with it any further after I got the bulb to reflash. But the key was in uploading the firmware. sudo esptool.py --port /dev/cu.usbserial --baud 115200 write_flash --flash_size 1MB 0x0 factory_firmware.bin. The key was --flash_size 1MB (must be after write_flash and before 0x0).