stelgenhof/AiLight #48: Losing config after powering off

nospam2k, 2018-12-20:
This doesn't happen every time but sometimes when I cycle power on the bulb, it goes back into the default AP state and I have to re-provision the bulb (connect to bulb AP, re-enter ssid, pw, etc). I have taken the .bin file from platformio and used "sudo esptool.py --port /dev/cu.usbserial -b 115200 write_flash 0x0 AiLight.bin" to flash it. Is there any reason not to do it this way?

stelgenhof, 2018-12-21:
I have not experienced myself that the config is lost after powering off. There is one bulb in our house that everybody uses with the regular mains switch and going strong for almost two years.

What version of the firmware are you using? You can definitely flash that way. Don't you use OTA?

nospam2k, 2018-12-23:
Ok, I'm still getting the above errors but it seems to be working. I haven't actually pulled the bulbs to check for losing config but I'm getting random results with compiling. I get this error every couple of compiles:

Assembler messages:
 Fatal error: can't create .pioenvs/prod-ota/libbc9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o: No such file or directory
 *** [.pioenvs/prod-ota/libbc9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o] Error 1
 Assembler messages:
 Fatal error: can't create .pioenvs/prod-ota/src/main.ino.cpp.o: No such file or directory
 *** [.pioenvs/prod-ota/src/main.ino.cpp.o] Error 1
 Assembler messages:

 Fatal error: can't create .pioenvs/prod-ota/libb84/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o: No such file or directory
 *** [.pioenvs/prod-ota/libb84/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o] Error 1

Also a couple of other random can't create xxxxxxx No such file or directory. When I click compile again, it works. This is on a Mac with Mojave.