stelgenhof/AiLight #49: Is there a wiki page for MQTT usage like REST

nospam2k, 2019-01-08:
I can't seem to find a page for using MQTT like the REST page. Ex. how to send/listen for a status request.

nospam2k, 2019-01-08:
Glad to hear it... but... I still don't know how to setup an mqtt to get status ;) I tried using mosquito with just the state topic and status topic and neither trigger a response (Ex. mosquitto_sub -h server -t /Lights/WifiLight1 or /Lights/WifiLight1/status).

brahmafear, 2019-01-09:
Hey nospam2k,

Your mosquito_sub command looks good. The topic to subscribe to will depend on your own configuration. The trick is that “/Light/...” is different than “Light/...”. 

Also, you can use 

mosquitto_sub -h server -v -R -t “#”

To subscribe to everything and see what topic c it is posted to.

nickrout, 2019-01-09:
Out of interest, what is the -R option? I cannot find it in man mosquitto_sub?

brahmafear, 2019-01-09:
> From the documentation:
>
> -R
> If this argument is given, messages that are received that have the retain bit set will not be printed. Messages with retain set are "stale", in that it is not known when they were originally published. When subscribing to a wildcard topic there may be a large number of retained messages. This argument suppresses their display.

Right:

-R basically says don’t show me all the old stuff. I use it when testing because I just want to see new messages coming in.

nospam2k, 2019-01-09:
Yes sorry for the typo. I am using Lights/wifi_light1. I even tried changing the state to Lights/wifi_light1/state. I cannot force a status message. If I send a set message, I get a state message but I cannot figure out how to get the state message without using a set. The status message doesn't seem to do anything.

brahmafear, 2019-01-09:
I don't believe there is a way to query state of device without changing
state of device via MQTT.  A few things might help:

The status topic is just for "online" or "offline".  It is used with MQTT's
last will feature to make an announcement on MQTT topic when the device
isn't connected to MQTT server.

You *could* receive a state update based on the last thing the bulb posted
to its state topic if you set the retain flag on the bulb settings and use
mosquitto_sub without the -R flag.  Since the bulb should post a new state
message every time something changes (via web, REST, or MQTT), that should
give the the current state of the bulb.  You'd also want to check status
topic to make sure bulb is online.

That all make sense?  These aren't really AiLight questions - more related
to how MQTT works.  I'm happy to continue to provide advice, but maybe
direct email me or check some of the MQTT primers on web.

nospam2k, 2019-01-09:
@brahmafear, I understand what you are saying... but, if the bulb experiences a power outage, how do I discover the state of the bulb? If the answer is "there is no way to query the state of the device without changing the state of the device", then I guess I need to switch to using REST, which is fine. I'm just wanting a solid yea or nay on whether the is an MQTT state request like REST.

nospam2k, 2019-01-09:
Ok, after digging through the code a bit, I found that if I send <my_set_topic> with a message of {status: true} I can get a state topic without having to actually change the state of the bulb because there is no STATUS command but since it is valid json, it passes the valid json test while passing back all the existing parameters to the MQTT message routine. Technically, you could send any valid non command json string (ex {"cmd": "status"}) and it would work. I'm just putting this here for anyone else interested in doing the same. Since my OP was in regards to an MQTT wiki page, I'll leave the closing of this to the moderator.

stelgenhof, 2019-01-10:
@nospam2k Posting a random/unknown but syntactically correct JSON payload that to the command topic can work indeed. However that is not by design. :)

Regarding to your questions on getting the state of the device, @brahmafear has perfectly explained it. Just the nature of a Publish/Subscribe pattern.
Technically it would be possible to allow for a command ('message') to get the current state of the device via MQTT, however that is something you shouldn't need when you use the retain option. 

The REST API implementation in the device is an HTTP server, whereas the MQTT part functions as a basic subscriber. Simply the REST API has the ability to get the state of the device, whereas the MQTT part has not.

As for the documentation, other Wiki-pages contain instructions how to change a setting (e.g. Color Temperature) for both the REST API as well as an example using `mosquitto_pub`. I believe most of the MQTT usage is covered in the documentation.

Or do you think something is still missing?( Except your question about retrieving the status via MQTT)

Cheers! Sacha

nospam2k, 2019-01-10:
Well, if the bulb experiences power loss, the state of the bulb becomes unknown since the Mqtt broker won't know the settings of the bulb. I've designed my own infrastructure using several open source projects to avoid "trusting" unknown servers (ex. Tasmota) and it's nice to have a fail safe means of knowing when something has happened so having a status topic for me is beneficial. Your code was fairly easy to follow so I'm happy with my work around. It would be trivial (from looking at your code) to add a command message to return the current state, but I'll leave that to you. Thanks for such a great firmware!

closed 2019-01-10 by nospam2k