stelgenhof/AiLight #51: "mqtt_json" platform no longer supported by Home Assistant

donkawechico, 2019-01-25:
AI-Light version: 0.4.1-dev
HA version: 0.86.2

As of HA 0.84, "mqtt_json" is no longer a valid platform:
https://github.com/home-assistant/home-assistant/pull/18227

Since upgrading, I've been unable to use AILight via Home Assistant, and see tons of spam in my logs saying:
[image]

Since the error doesn't explicitly suggest AILight is the source of the error, I did a recursive grep for "mqtt_json" in my HA directory and found nothing. Looking through AILight code, however, I see this line, which suggests the platfprm is being set to mqtt_json for discovery purposes:
https://github.com/stelgenhof/AiLight/blob/80666ddca444b8eb529b02ea17770ff10ff5b5b3/src/light.ino

It's certainly possible that I'm misinterpreting things here, but it feels to me like AILight firmware isn't accommodating the new APi for mqtt lights. 

Let me know if I can provide additional details.

stelgenhof, 2019-01-25:
@donkawechico Thanks for letting me know about this. I haven't upgraded HA myself so, haven't seen it yet. Let me have a look.

brahmafear, 2019-01-25:
Correct about change in HomeAssistant but your bulbs should still work (mine do). The mqtt_json platform is deprecated but still accepted. All that is needed in AiLight is a quick change in the discovery message sent. IIRC, the platform is changed from mqtt_json to mqtt and a new key schema=json is added.

stelgenhof, 2019-01-25:
From what I can tell from reading the HA PR in question, indeed the mqtt_json is still accepted. So the change should be straightforward. 

It will only mean that the AiLight firmware with this change will not work with earlier versions of HA (before 0.84).

SurfingSpider, 2019-01-25:
I just did the same change it just needs new documentation.

So in home assistant instead of putting ......

  - **platform: mqtt_json**
    name: 'Master Bedroom'
    state_topic: 'AiLight-2'
    command_topic: 'AiLight-2/set'
    color_temp: true
    brightness: true
    rgb: true
    white_value: true
    effect: false
    optimistic: false

you put.....

  - **platform: mqtt
    schema: json**
    name: 'Master Bedroom'
    state_topic: 'AiLight-2'
    command_topic: 'AiLight-2/set'
    color_temp: true
    brightness: true
    rgb: true
    white_value: true
    effect: false
    optimistic: false

stelgenhof, 2019-01-26:
@donkawechico As said, the change is simple but it will be a breaking change for those not running HA 0.84 yet with this update.

donkawechico, 2019-01-27:
Just curious -- does it necessarily need to break older versions? I'm envisioning something like `LEGACY_PLATFORM = true` in platform.io that switches the variable assignment in light.ino or something (I'm not familiar with the AiLight codebase so maybe that's all nonsense thinking).

I could try putting up a PR that makes that change, if that'd help. But if you already have a solution in the works, maybe not worth it?

donkawechico, 2019-01-28:
@stelgenhof Went ahead and whipped up a PR that (maybe) would resolve this issue with new discovery syntax. It builds locally, but I haven't had a chance to push it to an actual bulb for testing.

Could totally be a bad approach, but figured I'd at least give it a shot and see if it helps!

stelgenhof, 2019-01-29:
@donkawechico Thanks! See my comment on the PR

debsahu, 2019-02-05:
@stelgenhof See what we are doing here https://github.com/toblum/McLighting/blob/1578cafbf0aee2cbec9e5e1b76be3a6754072837/Arduino/McLighting/request_handlers.h#L1099

We have an issue open https://github.com/toblum/McLighting/issues/327 to address `on_command_type` in discovery message.

stelgenhof, 2019-02-05:
@debsahu Thanks for the update. Seems to me you are not following the specs :)

stelgenhof, 2019-02-06:
@donkawechico Just made changes that enable the new syntax for Home Assistant 0.84 (or later) and an option for users that are still on Home Assistant 0.84 (or older).

Compiled/tested the code successfully on my side. I only don't have HA < 0.84 so wasn't able to test that. However, the code changes are pretty minimal so the chances of any issues are very slim.

Cheers! Sacha

donkawechico, 2019-02-11:
Awesome! Thanks @stelgenhof!

Unrelated side-note, I work at a company under Microsoft, and at our recent hackathon, I presented a light-effects platform I've started that uses an external service + MQTT to generate effects off-bulb. Presented to 100 or so folks and spoke a lot about your firmware -- even brought in a flashed bulb to show it off. 

Just thought you might get a kick out of that :) Love these bulbs!

stelgenhof, 2019-02-12:
@donkawechico Thanks! That's so nice to hear :) Hope the audience enjoyed it.

Cheers! Sacha

stelgenhof, 2019-02-26:
@donkawechico Would you perhaps mind sharing an example of the light effects you demonstrated at that hackathon? It maybe makes a good practical example for others to see how to use this firmware :)

Cheers! Sacha

stelgenhof, 2019-04-08:
Fixed in the just released version 0.6.0.

closed 2019-04-08 by stelgenhof

donkawechico, 2019-04-09:
> @donkawechico Would you perhaps mind sharing an example of the light effects you demonstrated at that hackathon?

Happy to, but the code is reeeeally bad :P I was trying too hard to architect something extensible and ended up with a structural mess. I'm going to get back to the project soon, but not sure I'm ready to put my name to it yet ;)

That said, the mess is located here: https://github.com/donkawechico/mqtt-light-effects. Feel free to dig around (and even make suggestions if you like), but please ask before linking it in the wiki.

stelgenhof, 2019-04-10:
@donkawechico Thanks for sharing! Don't worry it is messy, still we can learn from it :) (No worries, I am not linking it anywhere).