stelgenhof/AiLight #54: Impressed all over again

nickrout, 2019-02-16:
I have been using AiLight for a while now, in fact I bought one of the original wooden jigs from @stelgenhof. 

Recently I wanted to upgrade to the latest code because of the changes since 0.84 of home assistant.

The laptop that I had when I first used AiLight had broken, and with it my install of platformio. Also my jig has a broken wire that will need a soldering iron. It was all looking to hard.

But no, I installed platformio cli on my headless NUC using pip. I cloned github of AiLight. I configured it and did a prod-ota install to the ip address of one of my bulbs. To my surprise it completed successfully, and I have now done them all. I say "surprise" not because I doubted the software, but because I doubted my ability to make it all work smoothly, and because I have never done an OTA update of any esp device previously. I had also never used platformio via the commandline.  

So I am very impressed at the ease with which it updated, even for a non dev like me. Congratulations on the system as a whole.

stelgenhof, 2019-02-16:
Thank you @nickrout for the kind words! Very happy to hear that. 

I take a lot of care and detail in creating software and want people that use it have a perfect experience: both in installing, setting up as well in using it. There are many projects/experiments out there that are simply released without proper testing / quality control :)

To be fair, it always is a bit of a (happy) surprise to me when the software works for everybody. There  are so many variables that are out of my control, making it hard to cover all scenarios. 

Lately the firmware hasn't seen many updates, as I think it slowly reaches the point of fulfillment. Also due to the limited memory of these light bulbs, it will be a challenge to add more features. I will soon do another release as there have been a couple of updates.

If there is anything else you'd like to see changed, added or fixed, just let me know.

Cheers! Sacha

darkfiberiru, 2019-03-27:
On the limited memory side how do you feel about configurable build for code sections to control memory usage especially for niche features,effects,whichever(don't build in what you don't need) Not expecting you to do it @stelgenhof  Just curious about what you think about that tatic.

stelgenhof, 2019-03-28:
@darkfiberiru Absolutely support that! If any feature/functionality can be configurable (within the limits of the memory) then perfect. 

One thing that is important is that certain functionality may need to have a UI, so that needs to be included as well. If a feature can be disabled, then the associated UI (if there is) needs to be disabled as well.

Cheers! Sacha

stelgenhof, 2019-06-09:
Closing this 'issue' :)

closed 2019-06-09 by stelgenhof