stelgenhof/AiLight #57: Building with VSCode + PlatformIO 4.0 extension gives error: "`.text' will not fit in region `iram1_0_seg'"

donkawechico, 2019-08-01:
**Issue Description:** 
After installing the PlatformIO extension for vscode (which apparently installs the latest PlatformIO 4.0 version), trying to run the "PlatformIO: Build" command fails with the error: 

> "`.text' will not fit in region `iram1_0_seg'"

**IDE:** VSCode with PlatformIO extension

**PlatformIO version:** 4.0

**Verbose Build output**

> Processing prod-ota (platform: espressif8266@1.5.0; board: esp8285; framework: arduino)
> ---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8285.html
> PLATFORM: Espressif 8266 1.5.0 > Generic ESP8285 Module
> HARDWARE: ESP8266 80MHz, 80KB RAM, 423.98KB Flash
> PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), tool-esptool 1.409.0 (4.9), framework-arduinoespressif8266 1.20300.1 (2.3.0)
> Converting main.ino
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 29 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <ArduinoJson> 5.13.4
> |-- <AsyncMqttClient> 0.8.2
> |   |-- <ESPAsyncTCP> 1.2.0
> |   |   |-- <ESP8266WiFi> 1.0
> |-- <my92xx> 3.0.3 #a876387
> |-- <ESPAsyncTCP> 1.2.0
> |   |-- <ESP8266WiFi> 1.0
> |-- <ESP Async WebServer> 1.2.2
> |   |-- <ESPAsyncTCP> 1.2.0
> |   |   |-- <ESP8266WiFi> 1.0
> |   |-- <Hash> 1.0
> |   |-- <ESP8266WiFi> 1.0
> |   |-- <ArduinoJson> 5.13.4
> |-- <ESP8266WiFi> 1.0
> |-- <ESP8266mDNS>
> |   |-- <ESP8266WiFi> 1.0
> |-- <Hash> 1.0
> |-- <AiLight>
> |   |-- <my92xx> 3.0.3 #a876387
> |-- <EEPROM> 1.0
> |-- <Ticker> 1.0
> |-- <ArduinoOTA> 1.0
> |   |-- <ESP8266mDNS>
> |   |   |-- <ESP8266WiFi> 1.0
> |   |-- <ESP8266WiFi> 1.0
> before_build([".pio/build/prod-ota/src/main.ino.cpp.o"], ["src/main.ino.cpp"])
> /Users/<username>/code/AiLight/node_modules/.bin/gulp
> Archiving .pio/build/prod-ota/lib78e/libArduinoJson_ID64.a
> Indexing .pio/build/prod-ota/lib78e/libArduinoJson_ID64.a
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/ESP8266WiFi.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/ESP8266WiFiAP.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/ESP8266WiFiScan.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/WiFiClient.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/WiFiClientSecure.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/WiFiServer.cpp.o
> Compiling .pio/build/prod-ota/libd25/ESP8266WiFi/WiFiUdp.cpp.o
> Compiling .pio/build/prod-ota/lib9ef/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o
> Compiling .pio/build/prod-ota/lib9ef/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o
> Compiling .pio/build/prod-ota/lib9ef/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o
> Compiling .pio/build/prod-ota/lib9ef/ESPAsyncTCP_ID305/SyncClient.cpp.o
> Compiling .pio/build/prod-ota/lib9ef/ESPAsyncTCP_ID305/tcp_axtls.c.o
> Compiling .pio/build/prod-ota/lib7a2/AsyncMqttClient_ID346/AsyncMqttClient.cpp.o
> Compiling .pio/build/prod-ota/lib7a2/AsyncMqttClient_ID346/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
[... 93 lines cut ...]
> Compiling .pio/build/prod-ota/FrameworkArduino/core_esp8266_wiring_pwm.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/core_esp8266_wiring_shift.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/debug.cpp.o
> Compiling .pio/build/prod-ota/FrameworkArduino/heap.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/libb64/cdecode.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/libb64/cencode.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/libc_replacements.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/pgmspace.cpp.o
> Compiling .pio/build/prod-ota/FrameworkArduino/setjmp.S.o
> Compiling .pio/build/prod-ota/FrameworkArduino/spiffs/spiffs_cache.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/spiffs/spiffs_check.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/spiffs/spiffs_gc.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/spiffs/spiffs_hydrogen.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/spiffs/spiffs_nucleus.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/spiffs_api.cpp.o
> Compiling .pio/build/prod-ota/FrameworkArduino/spiffs_hal.cpp.o
> Compiling .pio/build/prod-ota/FrameworkArduino/time.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/uart.c.o
> Compiling .pio/build/prod-ota/FrameworkArduino/umm_malloc/umm_malloc.c.o
> Archiving .pio/build/prod-ota/libFrameworkArduino.a
> Indexing .pio/build/prod-ota/libFrameworkArduino.a
> Linking .pio/build/prod-ota/firmware.elf
> /Users/<username>/.platformio/packages/toolchain-xtensa@1.40802.0/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/prod-ota/firmware.elf section `.text' will not fit in region `iram1_0_seg'
> collect2: error: ld returned 1 exit status
> *** [.pio/build/prod-ota/firmware.elf] Error 1

donkawechico, 2019-08-01:
FWIW, I was able to work around this by:
1. Opening VSCode
2. Cmnd+Shift+P -> PlatformIO: New Terminal
3. In terminal: `pip2.7 uninstall platformio`
4. `pip2.7 install platformio==3.6.7`
5. Rerun the OTA command: `platformio run --target upload --environment "prod-ota" --upload-port=<hostname>`

stelgenhof, 2019-08-01:
Thanks! I am aware of this (Just didn't create an issue yet). The issue is indeed with PlatformIO rather than this firmware. To my understanding a fix was made but perhaps not released, so kind of waiting for that.

Your workaround is indeed good: rollback PlatformIO back to a stable version. Perhaps this is something I should enforce in the project.

Cheers! Sacha

donkawechico, 2019-08-01:
Hm. As of a couple weeks ago, this is considered a "wontfix" by PlatformIO: https://github.com/platformio/platformio-core/issues/2780

Comment explaining why: https://github.com/arendst/Sonoff-Tasmota/issues/6073#issuecomment-511109803

I see Tasmota has made some changes very recently to platformio.ini to support 4.0:
https://github.com/arendst/Sonoff-Tasmota/commits/development/platformio.ini

Perhaps all that's needed is to add the "build_dir" parameter to "[platformio]" config section?

I can give it a shot locally and report back.

donkawechico, 2019-08-01:
Well by golly, it worked! 

Snippit:
```
[platformio]
default_envs = prod-ota
build_dir = .pioenvs
```

donkawechico, 2019-08-01:
https://github.com/stelgenhof/AiLight/pull/59

stelgenhof, 2019-08-02:
Thanks! Merged your PR. Compiled it and works perfectly!

closed 2019-08-02 by stelgenhof

aHcVolle, 2019-10-17:
FYI: ESP Async WebServer in version 1.2.3 breaks the build, with version 1.2.2 the build is still working fine

stelgenhof, 2019-10-18:
@aHcVolle Thanks for informing. Can you maybe create a separate issue for this?