stelgenhof/AiLight #62: Build fails with ESP Async Webserver version 1.2.3

aHcVolle, 2019-10-21:
**Issue Description:**
Latest version (master 0.6 or develop 0.6.1-dev) fails to build with 

> .pioenvs/dev/firmware.elf section `.text' will not fit in region `iram1_0_seg'

AiLight version: 0.6.x (maybe all?)

Toolchain:
espressif8266                   @ 1.5.0
tool-mkspiffs                   @ 1.102.0
tool-esptool                    @ 1.409.0
tool-espotapy                   @ 1.0.0 
framework-arduinoespressif8266  @ 1.20300.1
toolchain-xtensa                @ 1.40802.0

PlatformIO version: 4.0.3
OS: arch

When using ESP Async WebServer in version 1.2.2 the build odes succeed. 
I haven't tested the built file yet so i don't know if everything works fine but...

[build_log_server_1.2.3.log](https://github.com/stelgenhof/AiLight/files/3749257/build_log_server_1.2.3.log)
[build_log_server_1.2.2.log](https://github.com/stelgenhof/AiLight/files/3749258/build_log_server_1.2.2.log)

stelgenhof, 2019-10-21:
Thanks! There seems/was an issue with ESP Async WebServer: https://github.com/me-no-dev/ESPAsyncWebServer/issues/613

stelgenhof, 2019-11-04:
@aHcVolle As the ESP Async WebServer library still gives issues (no new release/tag available as of yet), I have locked the versions of the ESP* libraries to v1.2.2. These versions are know to work.

Going forward I only going to bump third party libraries if they are proven to work.

Can you pull the latest changes and give the compilation a try? You only need to update your platformio.ini file (See the example). Also you may need to delete the .pioenvs and/or piolibs directory in case you run into errors.

Cheers! Sacha

aHcVolle, 2019-11-04:
When is was testing around i did exactly the same as you did (I did only lock ESP Async WebServer to 1.2.2 but the other shouldn't make a problem).

I can run a test compile if you really want me to but i guess this shouldn't be neccessary.
Even if i did this i wouldn't test flash the file to my bulbs, i have massive ota problems and don't want to risk to brick them :)

stelgenhof, 2019-11-04:
@aHcVolle Don't need to compile if you already figured it out :)
Out of curiosity, your OTA problems are because of this firmware?

aHcVolle, 2019-11-04:
I have no idea from where my ota problems are coming...
Since this is the only platformio project I ever used I have no idea how to debug this thing.
I was using ailight 0.5.1-dev (somewhere around beginning of this year) where everything went super smooth.
When I wanted to update to the latest version (since I might have had wifi problems and the bulbs did switch to the blinking mode after a few hours) I'm getting a 100% flash when using OTA but the reported result is a random number (and the state is failed...). When i check the bulb, it most likely still has the old firmware on it. When I wanted to update my 4 bulbs yesterday I discovered that, if I flash like 20x in a row somehow the new firmware starts beeing used ....
I had 2 successful flashing attempts so there is no major problem, it seems to be timing related or whatever...

Btw, when i've got your attention :) 
I'm using openHAB as control and it's colorwheel selector is a little bit different that the one from homeassistent (it's using an array for the RGB values). I added some code to make it work, do you want me to create a PR to merge this?

stelgenhof, 2019-11-05:
Hmmm. I haven't experienced this myself. OTA can indeed be finicky.

As for OpenHAB, yes please! PR's are more than welcome. I'm not using OpenHAB so can't test it myself however.

stelgenhof, 2019-12-27:
@aHcVolle  The 'develop' branch contains new changes that addresses this issue with building the firmware. The code is not released yet but will be in the upcoming v0.7 version.
Issues with the build you experienced where coming from the external ESP8266/Arduino libraries.

If you have time, would appreciate it if you could give this branch a try.

Cheers! Sacha

aHcVolle, 2020-04-23:
I just wanted to update my devices (was on 0.61-dev before) and ota did work without any problems. Either this was already fixed in 0.61 or i had some network problems which somehow got resolved by now ....

closed 2020-04-23 by aHcVolle