stelgenhof/AiLight #64: espcomm_upload_mem failed

yeashen, 2019-11-26:
I connected my mac with AI-thinker LED, there are some fail when flashing the firmware,

what happend?  I found my AI-thinker flash is 25Q32xxx, the size is 32Mbit(8M Byte).

Processing dev (platform: espressif8266@1.5.0; board: esp8285; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=====     ]  48.5% (used 39716 bytes from 81920 bytes)
PROGRAM: [========  ]  84.1% (used 365164 bytes from 434160 bytes)
Configuring upload protocol...
Looking for upload port...
Auto-detected: /dev/cu.usbserial-1460
Uploading .pioenvs/dev/firmware.bin
Uploading 369312 bytes from .pioenvs/dev/firmware.bin to flash at 0x00000000
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed
*** [upload] Error 255
======================================================================================== [FAILED] Took 6.81 seconds ========================================================================================

Environment    Status    Duration
-------------  --------  ------------
dev            FAILED    00:00:06.809
dev-ota        IGNORED
prod           IGNORED
prod-ota       IGNORED
================================================================================== 1 failed, 0 succeeded in 00:00:06.809 ==================================================================================

stelgenhof, 2019-11-26:
This error can be caused by many different things. Wrongly connection, not enough voltage, etc. You may want to have a look at this older issue: https://github.com/esp8266/Arduino/issues/90. The comments give lots of information what may be wrong and how to solve it.

Cheers! Sacha

yeashen, 2019-11-26:
OK, I want to know if the difference in the size of flash has an effect?
your code adapts to 8Mbit flash, if my AI-thinker flash size is 32Mbit, 
Do I need to modify the platform.ini configure like this:
```shell
flag_flash_size = -Wl,-Tesp8266.flash.1m128.ld
```

yeashen, 2019-11-27:
Thanks for your  suggestion, I use Ubuntu OS run platformio run -t upload and at the same time press GPIO0 connect to GND, now It download OK.

closed 2019-11-27 by yeashen

stelgenhof, 2019-11-27:
Correct, to upload firmware the MCU GPIO0 pin needs to be grounded upon start.