stelgenhof/AiLight #65: OTA update over 0.5.1-dev not working on any device

nospam2k, 2019-12-13:
Not sure if it's just me, but I have 8 light bulbs with 0.5.1 that work fine but I can't upgrade to .6 no matter what I do. I've even tried making an OTA stub but the upload fails (as does .6) with various error codes. Sometimes 512, 1024, or longer various strings of numbers. I was able to flash over one bulb using the pins but it's a real hassle as my jig is very finicky and takes a few tries to get it work.

stelgenhof, 2019-12-13:
Could you maybe share a log file that shows some errors or other information?

Thanks! Sacha

nospam2k, 2019-12-13:
I gave up and reflashed all of them to .6. I'll leave you to close this if you want.

stelgenhof, 2019-12-13:
I would be still interested in the log file. OTA can be finicky so am keen to investigate this.

nospam2k, 2019-12-13:
I'm sorry I don't have any log files. I was going to try and setup a test using a 4m nodemcu. I did get an error on the OTA but it still uploaded and doing 2nd OTA was successful. I think being 4m helped so maybe a 1m chip test may show the problem. What ends up happening on the normal wifi light is it error and the original firmware is there but the settings are corrupted. They cannot be reset or changed from the web interface. I have attached the firmware bin if you want to test it.

[AiLight.bin.zip](https://github.com/stelgenhof/AiLight/files/3962121/AiLight.bin.zip)

stelgenhof, 2019-12-17:
Understood. Thank you for the file anyway. Not sure when, but will do some testing in the upcoming weeks :)

stelgenhof, 2019-12-24:
@nospam2k Just a small note: Have you tried to do an OTA with a different code/script (a simple blink for example). For OTA to work, the free space needs to equal or more than the current firmware/script. In case it is larger, OTA simply will not work. It may be possible the AiLight firmware is too large :(

nospam2k, 2019-12-24:
Yes. In my OP I said I tried making an "OTA stub".

stelgenhof, 2019-12-25:
@nospam2k Ah yes... you did :)

stelgenhof, 2019-12-25:
@nospam2k Did some tests with the binary you provided as well as my own code (0.5.1, 0.6, 0.6.1-dev) and was able to do an OTA upload without any issue.

[image]

Tested it on a lightbulb as well as on a ESP8266 board. As for the lightbulb, this had 0.5.1-dev installed and I uploaded the latest 0.6.1-dev code (like in your situation).

So not sure why yours fail to upload.

nospam2k, 2019-12-26:
I think, considering the version was older and not even a main branch, it's probably worth just closing this out.

stelgenhof, 2019-12-27:
Ok. Thanks! Anyway if you do have some issues again, just create another issue here :)

closed 2019-12-27 by stelgenhof