stelgenhof/AiLight #66: Mqtt commands don't work after Mosquito is restarted

nospam2k, 2019-12-19:
When I restart Mosquitto, the bulbs no longer respond to MQTT command until I restart the bulbs. Is there something I need to do to resync the bulbs with Mosquito? This is with the latest version of AiLight

stelgenhof, 2019-12-19:
Unfortunately this is indeed how it works. See for more details this issue: https://github.com/stelgenhof/AiLight/issues/56

stelgenhof, 2019-12-22:
@nospam2k I managed to do some preliminary changes that will check at regular intervals if the connection to the MQTT server is still present. 

If you like to try these out, please check the branch https://github.com/stelgenhof/AiLight/tree/feature/mqtt_reconnect and compile/upload this version.
Although I tested it and can confirm it works, be aware that issues may appear.

Cheers! Sacha

nospam2k, 2019-12-23:
Nice! I'll give it try when I have some free time.

stelgenhof, 2019-12-27:
@nospam2k  The 'develop' branch now contains the latest changes which you may want to use instead. The other `MQTT_Reconnect` branch has been merged into the '`develop`' branch. 

Cheers! Sacha

nospam2k, 2019-12-27:
I'll try it out after the holidays.

nospam2k, 2019-12-30:
Had some time today to try out current dev. Not sure what happened but MQTT isn't working at all. Messages are being passed but nothing happening on the light. Tried just ON and OFF and no response although the web interface is working fine. Reverted back to 0.6.1-dev. Working as before.

stelgenhof, 2019-12-30:
@nospam2k Can you change the **INIT_HASH** number in the `main.h` file to some other number? It is on line 72. (e.g change **0x5F** to **0x62**). The number itself doesn't matter as long as it is different.

This number is an internal check for the EEPROM configuration. If your EEPROM doesn't match the firmware code, the light may not work. By changing this code, it will enforce the new config to be initialized.

Thanks! Sacha

nospam2k, 2019-12-30:
Tried that. No change, same problem.

stelgenhof, 2020-01-02:
@nospam2k  Just made some changes in the 'develop' branch. I could replicate the issue that MQTT wasn't working. Basically a connection to the MQTT broker was never made in the code due to a change I made previously to the order when the connections (WiFi and MQTT) are made. I reverted that and should work now. I tested it on two different bulbs and a ESP8266 module.

donkawechico, 2020-02-03:
**1 week ago**: I completely switched out my home network hardware to Unisys stack, and moved home assistant to a new RPi4. AiBulbs (OLD firmware) appeared fine, but restarting was still making them lose connection, as expected.

**4 days ago**: I set up several WiFi networks (main, guest, IoT, and "NoT" (non-internet devices)). I assigned all bulbs to the "NoT" network. Bulbs continued to work fine.

**2-3 days ago**: I pushed the latest from `develop` branch to all bulbs. Tried restarting router, bulbs came back on their own and reconnected to MQTT server.  All seemed well!

**Today**: I logged into my router and found all the bulbs are now missing. I'm not aware of any router restarts or outages between yesterday and today.
-----
This could be due to the new networking hardware, or all the new ssids creating interference, or maybe the fact that my AP transmits at 22db, but my bulbs are all set to 1-3?

The only thing that makes me suspect the firmware still being the cause is that they all seemed to disappear at once. If it was just network interference, I would expect them to come in and out of existence.

I'll keep poking and update here.

stelgenhof, 2021-08-22:
Closing as considered resolved in v1.0.0

closed 2021-08-22 by stelgenhof