stelgenhof/AiLight #73: Using "transition" to turn off a bulb sets rgb to {0,0,0}. Turning back on results in no light.

donkawechico, 2020-03-27:
I'm trying to use `transition` to dim my lights until they're off. Each method I've tried has problems:

MQTT Payload | Result
------------- | -------------
`{ "state": "OFF", "transition": 5 }` | No dimming. Light just turns off after 5 seconds
`{ "brightness": 0, "transition": 5 }` | Dims nicely, but light state remains ON
`{ "state": "OFF", "brightness": 0, "transition": 5 }`| Dims nicely, light state is OFF, but rgb gets set to **{0,0,0}**. So when I turn back on the light, no light is produced regardless of brightness

Expectation would be for the color state to be retained after transition to brightness=0 and state=OFF.

donkawechico, 2020-03-27:
Just discovered that `{ "state": "ON", "transition": 5 }` also just turns on the light after 5 seconds rather than slowly brightening into its previous brightness.

I suppose my ideal behavior would be:

| MQTT Payload | Behavior (ex: initial brightness=150)
| ---------------- | -------------
| `{ "state": "OFF", "transition": 5 }` | Brightness decreases from 150 to 0 over 5 seconds, bulb state set to "off", prev brightness (150) saved internally
| `{ "state": "ON", "transition": 5 }` | Bulb state set to "on", brightness starts at 0 and goes to prev value (150) over 5 seconds. If no previous value stored, go from 0 to 255 by default.

Is this doable?

stelgenhof, 2020-03-31:
@donkawechico It has been a while since I implemented this and fit my needs back then. Nevertheless improvements can always be made :)
Definitely doable, just need to find a bit of time to work on that...

Cheers! Sacha

donkawechico, 2020-04-01:
Sure thing. I may take a stab at this as well. 

Speaking of which, is there a better way to do local dev testing than to upload the firmware to an actual bulb and seeing if it works? Maybe some kind of arduino emulator?

donkawechico, 2020-09-07:
I've started taking a look at this again, btw.

I tried changing it to gradually decrease _brightness_ to 0 instead of decreasing _color_ to {0,0,0}, but this just creates the same problem in a different place -- that is, brightness transitions to "0", so when I turn back to ON, the bulb will still be brightness of 0.

Here's a Draft PR where you can follow the progress: https://github.com/stelgenhof/AiLight/pull/80

stelgenhof, 2020-09-10:
Thanks! @donkawechico Let me have a look at it tomorrow.

donkawechico, 2020-09-10:
> Thanks! @donkawechico Let me have a look at it tomorrow.

Sure thing, but not much there at the moment. Still need to think of a way to restore previous brightness

closed 2020-11-21 by github-actions[bot]