stelgenhof/AiLight #75: MQTT discovery broken since HA 0.108

nickrout, 2020-04-10:
I am running Version 0.6-dev pf AiLight and recently moved to 0.108 of home assistant. My lights no longer work. I suspect this https://github.com/home-assistant/core/pull/32909 

The discovery packet sent from the bulb is
```
homeassistant/light/foyerlight2/config {"name":"foyerlight2","platform":"mqtt_json","state_topic":"foyerlight2","command_topic":"foyerlight2/set","rgb":true,"color_temp":true,"brightness":true,"white_value":true,"availability_topic":"foyerlight2/status"}
```

It may be that this is fixed in a later version of AiLight?

stelgenhof, 2020-04-10:
Oh! Thanks for letting me know. I haven't looked at HomeAssistant for a while so must have missed that. I will have a look to fix that.

nickrout, 2020-04-11:
Looking further, it looks like my bulb firmware must been compiled with MQTT_HOMEASSISTANT_DISCOVERY_PRE_0_84 set true. Looking at src/light.ino and the setup of md_root

stelgenhof, 2020-04-13:
@nickrout Have you tried recompiling with MQTT_HOMEASSISTANT_DISCOVERY_PRE_0_84 flag set to `FALSE`?

nickrout, 2020-04-15:
My linux laptop with platformio had a collision with the contents of a wineglass and I haven't replicated a platformio setup yet. I'll get on to it.

nickrout, 2020-04-15:
@stelgenhof how do I know whether my bulb has OTA enabled, in the absence of the .ini files I compiled with? The bulbs don't respond to a telnet to post 8266 and nmap only shows one  port open - 80.

stelgenhof, 2020-04-15:
@nickrout I don't know any way to know that. Telnet will not work as the firmware doesn't run a telnet server. I had the same issue once and just reflashed the firmware directly.

closed 2020-11-06 by github-actions[bot]

domeOo, 2021-01-09:
I think I have the same issue.